dokutan/mouse_m908 #1: Reverser engenieering USB protocol.

thegala, 2020-04-22:
Hello, 
Since I have no other means to contact you I opened this issue.
I bought myself, a Redragon M709. Because I can't change any mouse related options on Linux I want write my own app like yours. I am interested how'd you reverse engineered protocol.

Thx.

dokutan, 2020-04-22:
Opening this issue is a perfectly fine way to get in touch.

Here are some links that should help get you started:
[https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/](https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/)
[https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol)
[https://github.com/dokutan/rgb_keyboard/tree/master/doc](https://github.com/dokutan/rgb_keyboard/tree/master/doc) (from my other usb reverse engineering project)

Before starting the reverse engineering, you could try if this software is (partially) compatible:
1. get the USB vendor id (VID) and product id (PID) of your mouse by running ``lsusb``
2. If the VID is 0x0c45 and PID is 0xfc4d then there is a good chance of compatibility (identical to the M908). If the PID is different you need to specify the bus and device address (from lsusb) when running this software (this skips the VID and PID check).
3. Even if the software does not work without modification, you can look at the source code to find out if the protocol is similar or identical, and the difference is the way data is sent.

In any case it would be really great if you could report your findings regarding the compatibility. If it turns out that there is a way of making this software compatible, and you are interested in that, i would be happy to help.

I am leaving this issue open for now, so that you can report back whatever you find, or if you have any other questions. I hope this helps.

thegala, 2020-04-29:
Thanks.  Sry for late answer. I am currently writing my thesis, so when I got some time I will gave this more serious look.
No, VID and PID are different.  _04d9:fc2a_
I already did some USB data capture on Windows machine. It looks that protocol is very similar to your protocol.  I will give it more detail look when I got more time.

dokutan, 2020-04-29:
I am glad the usb capture worked out. I have added an annotated version of my captured data in the documentation directory in case you want to have a look at it. 

Also i have given you a wrong VID in my previous answer, the correct pair is 04d9:fc4d, which means these mice are indeed from the same manufacturer.
If you send me your captured data i might be able to add initial support.

Thanks for your work and i wish you all the best with your thesis.

thegala, 2020-07-11:
Finally I did capture data!!
Sry for sooooo late response.

[m709_1.zip](https://github.com/dokutan/mouse_m908/files/4907765/m709_1.zip)
[m709_2.zip](https://github.com/dokutan/mouse_m908/files/4907766/m709_2.zip)

dokutan, 2020-07-12:
I had a look at the captured data, changing the profile is identical to the M908, and the rest appear to be very similar as well.
However there is one major difference, each usb packet is repeated twice, i do not know whether it is required to send each packet twice or not.

I have added a new branch m709, where the usb pid is changed, it would be great if you could try to to change the profile and read settings with that version, please note that you might have to run the program with ``sudo``.

If reading the settings works, writing should work as well, if not it would be useful to capture the communication when starting the official software.

thegala, 2020-07-12:
> However there is one major difference, each usb packet is repeated twice, i do not know whether it is required to send each packet twice or not.

Yes, I saw that maybe is some kind of bug but I don't think so.

Thx once again. I will do that, probably in next few days.

thegala, 2020-07-12:
> If reading the settings works, writing should work as well, if not it would be useful to capture the communication when starting the official software.
Please can you tell me how did you record mouse reading usb packets?
I can't seem to force Redragon application to read from mouse.

dokutan, 2020-07-12:
You need to completely close the Redragon application, start a capture and then open the Redragon application again. Maybe look in the Taskmanager to make sure it is not running in the background.

thegala, 2020-07-12:
> You need to completely close the Redragon application, start a capture and then open the Redragon application again. Maybe look in the Taskmanager to make sure it is not running in the background.
> 
> Does changing the profile work with the updated version?

I will give it a try. I was trying that closing then starting it again.  Maybe it was runing in background.
If you mean in Redragon application yes.

dokutan, 2020-07-12:
Sorry, i meant the version from the ``m709`` branch. Testing this version should tell whether the duplicated packets are neccessary.

thegala, 2020-07-12:
> Sorry, i meant the version from the `m709` branch. Testing this version should tell whether the duplicated packets are neccessary.

It works.

dokutan, 2020-07-12:
That is good to hear, i just finished implementing everything required to have support for multiple mice in the ``multiple-models`` branch.
All that is left now is to implement the  M709 specific features, for now you could try to:
- capture more data (reading the settings, maybe sending a macro)
- build from the ``multiple-models`` branch and run ``mouse_m908 -M709 -D -`` and ``mouse_m908 -M709 -R -`` to test reading the settings

I will get back to you when i have implemented sending the config to the mouse.

dokutan, 2020-07-12:
I have added support for sending settings to the M709 in the ``multiple-models`` branch, does
```
mouse_m908 -M709 -c example709.ini
```
work (correctly)?

TODO:
- Test sending settings
- Sending macros, requires testing
- Reading settings, requires testing and most likely usb capture

thegala, 2020-07-12:
> You need to completely close the Redragon application, start a capture and then open the Redragon application again. Maybe look in the Taskmanager to make sure it is not running in the background.

No luck. Still I didn't got any data. Any suggestion?

thegala, 2020-07-12:
> I have added support for sending settings to the M709 in the `multiple-models` branch, does
>
> ```
> mouse_m908 -M709 -c example709.ini
> ```
>
> work (correctly)?
>
> TODO:
>
>     * Test sending settings
>
>     * [...]

Yes it's working. I would test setting and macros.
Btw u are fast. I started working on M709 support but u where quicker.

dokutan, 2020-07-13:
Sorry, the reading capture does not seem to contain any useful data, maybe the M709 deos not support that feature? Could you post the output of ``mouse_m908 -M709 -D -`` and ``mouse_m908 -M709 -R -`` if there is any?

I would suggest to test the other features for now, and to get them working if they are not (changing settings, macros).

Also, if you have changed the code, pull requests are welcome.