dokutan/mouse_m908 #2: Reverse engeneering USB protocol (Redragon M711).

rolimans, 2020-06-11:
Hello, I'd like to the same you did, but with the model M711.
This is my first attempt reverse engineering something, so I'm having a hard time starting it.
After setting up the VM and Wireshark, I was able to capture some packets after changing values in the original software, but all packages were very small (average of 80 bytes) and were nothing like the dump found in your file "documentation/default-annotated".
Am I doing something wrong, or are the packages meant to be this size, and if so, how should I concat them to a better understending of the protocol.

dokutan, 2020-06-11:
I'm glad that the README helped, if you have any other questions feel free to ask.

I would recommend starting by capturing the data and writing a simple program to open the mouse, send the captured data, and close the mouse. After that works, the main work is capturing data for different settings, running ``diff`` on the captures and changing the correct values in the data.

Would you prefer to add support for the M711 to this program, or to have a separate program?
I am open to any pull requests, and if you want to add support to this program i would look in to the required changes to support multiple mice.

dokutan, 2020-07-31:
I have added experimental support for the M711, there are still a few features missing because i don't have any captured data for them, and nothing is tested. I am leaving this issue open for now, in case you want to try it out and report what is broken and what works.

rolimans, 2020-07-31:
I'm sorry for taking so long to answer, I've made quite a nice progress, and I'd be willing add support for the M711 to your program, or to have a separate program. It really depends a lot on what is best. I will checkout the support you have added and comment and collaborate with it. The only problem is that I am very busy nowadays because of work and college, so for now there is not much I can do. But in the future be assured to hear back from me.

dokutan, 2020-07-31:
As there already is initial support for the M711 (and others) in this program, i think it is the best (and easiest) to have support for these mice in one program, and would be glad about any contribution and collaboration towards better hardware support. Feel free to take your time, and whenever you are ready, open a pull request.
Furthermore if you have any initial code or captured data, feel free to share it, maybe i can use it to implement the missing features if you currently lack the time to do so.

nwildner, 2021-02-08:
Hey @dokutan I bought a Mouse Cobra M711. Is there any data capture/commands that i could run to help on improving this device support?

lsusb, grepping relevant id:

`Bus 003 Device 005: ID 04d9:fc30 Holtek Semiconductor, Inc. `

dmesg, cropping relevant usb mouse stuff:

```
dmesg | grep -i mouse

[    3.317097] usb 3-2.4: Product: USB Gaming Mouse
[    3.330125] input: USB Gaming Mouse as /devices/pci0000:00/0000:00:1b.4/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:02.0/0000:09:00.0/usb3/3-2/3-2.4/3-2.4:1.0/0003:04D9:FC30.0003/input/input13
[    3.330183] hid-generic 0003:04D9:FC30.0003: input,hidraw2: USB HID v1.10 Mouse [USB Gaming Mouse] on usb-0000:09:00.0-2.4/input0
[    3.332778] input: USB Gaming Mouse as /devices/pci0000:00/0000:00:1b.4/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:02.0/0000:09:00.0/usb3/3-2/3-2.4/3-2.4:1.1/0003:04D9:FC30.0004/input/input14
[    3.384792] hid-generic 0003:04D9:FC30.0004: input,hidraw3: USB HID v1.10 Keyboard [USB Gaming Mouse] on usb-0000:09:00.0-2.4/input1
[    3.390123] input: USB Gaming Mouse Consumer Control as /devices/pci0000:00/0000:00:1b.4/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:02.0/0000:09:00.0/usb3/3-2/3-2.4/3-2.4:1.2/0003:04D9:FC30.0005/input/input15
[    3.441753] input: USB Gaming Mouse as /devices/pci0000:00/0000:00:1b.4/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:02.0/0000:09:00.0/usb3/3-2/3-2.4/3-2.4:1.2/0003:04D9:FC30.0005/input/input16
[    3.441825] hid-generic 0003:04D9:FC30.0005: input,hiddev97,hidraw4: USB HID v1.10 Device [USB Gaming Mouse] on usb-0000:09:00.0-2.4/input2
```

dokutan, 2021-02-08:
Yes, these are the things that you can try:
1. To get a first impression about the capabilities of the mouse, run ``mouse_m908 -R -`` and ``mouse_m908 -D -``
2. Test the features that are already implemented (see the list in the README), for features that don't work or are not implemented for thew M711 try the generic backend with the ``-M generic`` option. I strongly recommend that you have the official software available (in a VM) because there is a small chance that changing the settings could make the mouse unusable (e.g. no movement in one direction).
3. If the previous steps were not sufficient to implement all features, you can capture the USB data sent by the official software with wireshark.

A few links that i copied from a previous issue about the capturing process:
https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc

That should be all, just ask if there are any questions remaining.

rolimans, 2021-02-08:
I've had some progress, but had to leave this project on hold for a while because of work and univertsity. A lot of settings from the m908 work right away with the m711 (same config bytes). I have some notes from when I was reverse engineering it myself, but I can barely understand them, so it probably won't be of much help.

dokutan, 2021-02-08:
No problem, that a lot of settings work gives me hope that it shouldn't be to dificult to implement the differing features (mainly different buttons i would assume), if you could post your notes i would have a look, we don't know if they are helpful unless i try.

rolimans, 2021-02-08:
```
dpi_enable  ok
dpi ok
lighmode ok

report rate ok
brighness ok
color ok

change dpi?
scroll speed?

speed:

    breathing,rainbow:
        low 08
        med 05
        high 02
    flashing:
        low 08
        med 05
        high 01
    alternating,reactive:
        default 8
    static,off:
        default 0

```
~~int mouse_m908::read_settings(){  // Why?~~

~~check windows only and clicks in macros~~

These are the notes, they are confusing as I said.
The numbers aside the rgb modes are the needed bytes to be sent for each mode (if I'm not mistaken). I just cant remember the position of those bytes.

rolimans, 2021-02-08:
Checking the diffs I can see a big addition in `data.cpp`. I can't remember if I did those but it seems so.
The code follows:

```
std::map<std::string, uint8_t> mouse_m908::_c_windows_only_actions = {
	{"basic_cut", 0x11},
	{"basic_copy", 0x12},
	{"basic_paste", 0x13},
	{"basic_select_all", 0x14},
	{"basic_find", 0x15},
	{"basic_new", 0x16},
	{"basic_print", 0x17},
	{"basic_save", 0x18},
	{"advanced_switch_window", 0x19},
	{"advanced_close_window", 0x1a},
	{"advanced_open_explorer", 0x1b},
	{"advanced_run", 0x1c},
	{"advanced_show_desktop", 0x1d},
	{"advanced_lock_pcme", 0x1e},
	{"advanced_browser_home", 0x1f},
	{"advanced_browser_backward", 0x20},
	{"advanced_browser_forward", 0x21},
	{"advanced_browser_stop", 0x22},
	{"advanced_browser_refresh", 0x23},
	{"advanced_browser_search", 0x24},
	{"advanced_browser_favorite", 0x25},
	{"advanced_mail", 0x26}};
```

dokutan, 2021-02-08:
Thanks, that was quite helpful. A bit of information for reference: the LED modes seem to be identical, i have since added a [description of the byte values](https://github.com/dokutan/mouse_m908/tree/master/documentation#led-modes), the position can be found in the source or the [annotated capture](https://github.com/dokutan/mouse_m908/blob/master/documentation/default-annotated_m908.pdf). The windows only actions aren't handled by windows but rely on the official software to be running in the background, therefore i wouldn't bother implementing them (unless someone wants them).

If i understand it correctly, most settings did work, however in the recent versions of the M711 backend changing the scrollspeed, DPI, button mapping and USB poll/report rate were disabled due to a lack of information. I have enabled the generic implementation of all settings in the M711 backend in 40f42ad0ef56b45edf831057d3d5c75ac8a76100, if you @rolimans or @nwildner were to confirm that (some of) changing these settings works now, i would update the documentation accordingly.

nwildner, 2021-02-09:
Ok. Maybe i'm stupid(and if i'm being, dont be afraid to tell me..lol) but i've tried to change simple things on my config file to test led colors, and setting them for each profile. Example:

```
[profile1]

# LED settings
color=ff0000
brightness=2
speed=0
lightmode=on
report_rate=500
--snip--

```

```
[profile2]

# LED settings
color=00ff00
brightness=2
speed=2
#lightmode=wave
lightmode=on
report_rate=500

```

So, profile 1 is red, while profile 2 is blue(should be green). Both profiles change the led color to white 5 seconds right after i execute something like `sudo mouse_m908 -c mouse.ini -p 1`

dokutan, 2021-02-09:
That means that sending the settings to the mouse works (white is the default value for this software) but there is a bug in the  preparation of the data packet (the offsets for the color, brightness and LED speed were wrong), that should be fixed in 040c40141b808dc99eb16c9dba7346de2b2630a9.

PS @nwildner: I just noticed your fork with the PKBUILD is gone (i am using Arch myself and was therefore interested), are there any problems?

nwildner, 2021-02-10:
> That means that sending the settings to the mouse works (white is the default value for this software) but there is a bug in the preparation of the data packet (the offsets for the color, brightness and LED speed were wrong), that should be fixed in [040c401](https://github.com/dokutan/mouse_m908/commit/040c40141b808dc99eb16c9dba7346de2b2630a9).
> 
> PS @nwildner: I just noticed your fork with the PKBUILD is gone (i am using Arch myself and was therefore interested), are there any problems?

I'll be ready to do extensive testing after thursday, then i'll finish that pkgbuild :)

matteoturini, 2021-05-31:
I have an M711 and here if you want testing or help. Let me know what to do!

matteoturini, 2021-05-31:
``ìni
# Macros

;## macro8
;# unknown, please report as bug: ff ff 0
```

matteoturini, 2021-05-31:
What are these hex values? They are 100% not my DPI.
```
# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=0x0b00
dpi2_enable=1
dpi2=0x1600
dpi3_enable=1
dpi3=0x1400
dpi4_enable=1
dpi4=0x4500
dpi5_enable=1
dpi5=0x7300
```

dokutan, 2021-05-31:
There are four main aspects that need varying levels of effort. For the most part it boils down to trying various settings and checking if the results match as a big part of the model specific code has already been written. From my experience i can strongly recommend that you have the official software and a second input device available before trying to change the settings on the mouse (see https://github.com/dokutan/mouse_m908#safety).
- I currently do not know if the LED related settings (lightmode, brightness, color, speed) can be read and changed correctly. To confirm this check if the output of ``mouse_m908 -R -`` is correct and try changing these settings to a few different values.
-  The same is true for the button mapping, however due to the differences between the mice there are three types of failure that could occur:
    - The number of buttons assumed by this software is wrong
    - The buttons are swapped, e.g. button_left is interpreted as button_right
    - The buttons are labeled incorrectly
- To check the macros use the same procedure. Please note that some mice have less than 15 macros available.
- To address the DPI values, the hex values you see is part of the raw bytecode sent by/to the mouse and map to actual human-readable DPI values. Unfortunately these mappings vary between mice and follow no known pattern that is reasonably usable, therefore to be able to use the actual DPI values you would need to use the official software to set each of the available values, then look at the output from mouse_m908 for the matching hex values. Given that information it is relatively trivial to fix this issue.

If anything doesn't work as it should, please attach the config that you used and the output of ``mouse_m908 -R -`` and ``mouse_m908 -D -``. Telling me that something works is equally valuable because i can then update the README.md.

matteoturini, 2021-06-01:
Found a pattern (I think?)
[image]
[image]

matteoturini, 2021-06-01:
isn't it weird that 100dpi = 0x0200? 200dpi = 0x400?

dokutan, 2021-06-01:
> Found a pattern (I think?)

Yes, these values follow a basic pattern (here are the values from the [M908](https://github.com/dokutan/mouse_m908/blob/3db937dba3046c1bc08dc027f1a7e3bd92ef626d/include/m908/data.cpp#L55), and for the [M719](https://github.com/dokutan/mouse_m908/blob/3db937dba3046c1bc08dc027f1a7e3bd92ef626d/include/m719/data.cpp#L45)):
- if the DPI value is increased by 100 the first byte gets increased by 2 _or_ 3
-  at some point for larger DPI, the step size between valid values changes from 100 to 200 and the second byte gets changed from 0 to 1 (the whole DPI value could be seen as a little-endian word, but this change doesn't happen at 0xff 0x00)

However, because the difference between adjacent values varies and the point at which the step size changes is different for each mouse i don't see a way around trying out at least the majority of DPI values. Additionally the minimum and maximum DPI (and therefore the amount of valid values) vary between mice.

> isn't it weird that 100dpi = 0x0200? 200dpi = 0x400?

I personally wouldn't interpret these values as anything but arbitrary keys for a map, therefore no.

LeonardHM, 2021-09-15:
I didn't figure out how to use it, now my mouse doesn't move vertically.  Would you help me?

dokutan, 2021-09-15:
@LeonardHM I suggest you try out the official software to fix your mouse. To debug this program it would be helpful if you could provide the config you used, the output of ``mouse_m908 -R -`` and ``mouse_m908 -D -``, the version (which release or did you build from the master branch?) and if all profiles and DPI levels are affected.

LeonardHM, 2021-09-15:
I installed libusb, cloned the repository and installed it. I only use command line. I just used the commands from the repository page with some template overrides (examples/example_m711.ini -M 711). I probably did something wrong

$ mouse_m908 -R -
# Model: 711
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# This feature is currently untested, please report your results.
# Note: reading the scrollspeed is not supported by the mouse.

# Currently active profile: 1

[profile1]

# LED settings
color=50ff00
brightness=2
speed=2
lightmode=static
report_rate=500

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=0
dpi1=200
dpi2_enable=1
dpi2=0x1600
dpi3_enable=1
dpi3=0x2d00
dpi4_enable=1
dpi4=0x4300
dpi5_enable=0
dpi5=0x8c00

# Button mapping is not implemented due to a lack of captured data

[profile2]

# LED settings
color=5000ff
brightness=3
speed=4
lightmode=breathing
report_rate=125

# DPI settings
# Active dpi level for this profile: 5
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=0x1600
dpi3_enable=1
dpi3=0x2d00
dpi4_enable=1
dpi4=0x4300
dpi5_enable=1
dpi5=0x8c00

# Button mapping is not implemented due to a lack of captured data

[profile3]

# LED settings
color=ffffff
brightness=3
speed=2
lightmode=static
report_rate=125

# DPI settings
# Active dpi level for this profile: 2
[... 325 lines cut ...]
02 08 b6 62 04 00 fa fa 8a 00 00 00 

02 08 ba 62 04 00 fa fa 90 00 e0 00 

02 08 be 62 04 00 fa fa 90 00 e1 00 

02 08 ca 62 04 00 fa fa 8b 00 00 00 

02 08 52 63 04 00 fa fa 81 00 00 00 

02 08 56 63 04 00 fa fa 82 00 00 00 

02 08 5a 63 04 00 fa fa 83 00 00 00 

02 08 5e 63 04 00 fa fa 85 00 00 00 

02 08 62 63 04 00 fa fa 88 00 00 00 

02 08 66 63 04 00 fa fa 8a 00 00 00 

02 08 6a 63 04 00 fa fa 85 00 00 00 

02 08 6e 63 04 00 fa fa 84 00 00 00 

02 08 7a 63 04 00 fa fa 8b 00 00 00

dokutan, 2021-09-16:
> I probably did something wrong

Sorry that i am only replying now, as far as i can tell you did absolutely nothing wrong.

After having a look at the data you provided i see no obvious cause for this issue, i suspect it might be caused by accidentally setting the DPI in the vertical direction to 0.
To implement a proper fix would require capturing the usb communication with the official software when changing the DPI, have a look at [this](https://github.com/dokutan/mouse_m908/issues/2#issuecomment-775276491) previous comment for details. If that is not possible i _might_ be able to figure something out if you reset the mouse with the official software and then attach the output of running:
```
mouse_m908 -D -
mouse_m908 -R config.ini
mouse_m908 -c config.ini
mouse_m908 -D -
```

LeonardHM, 2021-09-17:
I used the two redragon programs for this model and it didn't work.  I think the only way is trying to modify your code

dokutan, 2021-09-18:
> I think the only way is trying to modify your code

That might be true, however there are two problems with that approach:
1. That even the official software is broken makes it look like a bug in the firmware to me, which could mean that even knowing what exactly to modify would not lead to success.
2. I don't have a clue which commands to modify without a capture of the data sent by official software.

LeonardHM, 2021-09-30:
> That might be true, however there are two problems with that approach:
>
> 1. That even the official software is broken makes it look like a bug in the firmware to me, which could mean that even [...]

I managed to reset.  What do you need to try to fix the code?

dokutan, 2021-10-01:
> What do you need to try to fix the code?

To get started with the main issue: a capture of applying the default settings with the official software. Additionally a second capture where you change a single DPI value and a screenshot showing the settings would be helpful.

> I managed to reset.

I am genuinely curious how, if you could give me a brief description i would add it to the README.md in case someone else encounters a similar problem.

LeonardHM, 2021-10-01:
How do I capture?  I clone the repository, and do "mouse_m908 -R -" and "mouse_m908 -D -" ?

To reset, it was (simply) using the official program [this](https://solidpower.cloudstorage.pw/Software/Redragon/Mouse/M711-FPS/Redragon%20King%20Cobra%20M711-FPS%20%28EN%20-%20BETA%29.exe)  but on Windows 10. On 7 it didn't work, maybe due to machine problem or incompatibility, I don't know.

dokutan, 2021-10-01:
> How do I capture?

You can use wireshark, here are a few links copied from a previous reply:
https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc

Please ask if anything remains unclear.

LeonardHM, 2021-10-01:
Wow. This sounds a little complicated.  But I found some tutorials in my language, I will learn a little and do it soon.  Is there any place we can get in touch?  I believe that it is getting very polluted here and it will make it difficult for anyone to access looking for information

dokutan, 2021-10-01:
> Is there any place we can get in touch?

Sure, just send me an email at ``dokutan <at> tutanota <dot> com``

dokutan, 2021-10-07:
@rasmysamy Thanks, as it turns out there isn't even a difference between the captures. Where other mice use two bytes for the DPI the M711 uses four, i.e. two bytes per axis. The problem was that the last two bytes were always set to 0 by this software.

I have implemented a simple fix (setting y-dpi = x-dpi = dpi) in 3d684797c080c6aa33118b91391211dad12a0503, and experimental support for different x- and y-dpi in https://github.com/dokutan/mouse_m908/tree/m711-xy-dpi. I would merge that when someone confirms that it works.

rasmysamy, 2021-10-07:
@dokutan I'll be able to test the fix today or tomorrow, assuming the build
works without an issue. Thanks, and have a nice day!

1nekomata, 2021-11-13:
Hello. I have the Redragon M711 Cobra mouse and have captured some packets while using the mouse and applying a few settings in the official software. Should i send them here or open a new Issue?

dokutan, 2021-11-15:
@1nekomata Feel free to post them here.

1nekomata, 2021-11-19:
@dokutan Sorry for the late reply. Here is the [file](https://github.com/dokutan/mouse_m908/files/7572955/1.zip)

1nekomata, 2021-11-19:
Let me know if you would need something else

dokutan, 2021-11-22:
@1nekomata The most useful thing would be to try out as many features as possible, all features are implemented in some form but are not tested. Even reporting that a feature works as expected is useful, as that would allow me to update the readme.

1nekomata, 2021-11-23:
Ok, i'll do that tomorrow

1nekomata, 2021-11-24:
@dokutan Can you tell me what i should try applying?

dokutan, 2021-11-24:
@1nekomata You can use the example config and then set arbitrary values for the individual settings. Only when something doesn't work the values you used become interesting.

1nekomata, 2021-11-24:
@dokutan i'll start with dpi then. 0200 is 100 dpi 0400 is 200 dpi and so on right?

1nekomata, 2021-11-24:
[default config coming from windows](https://github.com/dokutan/mouse_m908/files/7598438/cobra711.txt)

1nekomata, 2021-11-24:
@dokutan i applied [this.](https://github.com/dokutan/mouse_m908/files/7598470/cobra711test.txt) color works, dpi works, need to remap a few buttons (dpi cycle is mapped to page back, change color mode isn't mapped, page forward isn't mapped, dpi lower isn't mapped) and applying polling rate works

1nekomata, 2021-11-24:
@dokutan I'm gonna continue tomorrow and will send a config with all keys mapped properly and all color modes tested

1nekomata, 2021-11-24:
so far static and wave color mode works fine.

1nekomata, 2021-11-25:
@dokutan applying dpi works. disabling dpi modes works too. changing dpi works too. all color modes (seem to) works. only buttons that are not working are
[image]
1 isn't working, 2 cycles dpi (should be backwards), 3 switches dpi up (works normally), 4 isn't working (should be dpi down), 5 is backwards (should be cycle color mode)

1nekomata, 2021-11-25:
dokutan we don't have enough keys to be mapped to the mouse (8 vs 9) so the key i marked as 4 in the image above is the same key as the key i marked as 1

1nekomata, 2021-11-25:
@dokutan here is the pcap of [button1](https://github.com/dokutan/mouse_m908/files/7603484/button1.pcapng.zip
and [button4](https://github.com/dokutan/mouse_m908/files/7603485/button4.pcapng.zip)
)

dokutan, 2021-11-25:
@1nekomata Thanks, your information and captured data proved to be quite useful. I have added support for remapping all buttions in f04a4cc5207788c4c6679f7e67f054275fb0067c. The example has been updated with a full list of buttons so you can try it out easily.

If that works correctly, the only missing info is the mapping between actual DPI values and the raw bytes. To get that you can set the DPI using the official software, then run ``mouse_m908 -R -`` and repeat the process until you have covered all possible values (100-500 are already known and can be skipped). If the button mapping doesn't work i would have another look at that.

1nekomata, 2021-11-25:
ok i will do that. i am setting up a VM and am gonna do that ASAP

1nekomata, 2021-11-27:
@dokutan 
dpi1=0x0d000000 = dpi 600
dpi2=0x0f000000 = dpi 700
dpi3=0x12000000 = dpi 800
dpi4=0x14000000 = dpi 900
dpi5=0x16000000 = dpi 1000
sadly i haven't found a way to refresh RDCfg on Windows to see what dpi i just applied on linux

1nekomata, 2021-11-27:
@dokutan would mouse_m908 work on Win10?

1nekomata, 2021-11-27:
@dokutan all mouse buttons work as expected now thanks a lot!

dokutan, 2021-11-27:
@1nekomata Thanks for the DPI mapping. The known values are identical to the M908, if you can tell me the maximum supported DPI i could copy at least some values from the M908. 

> all mouse buttons work as expected now thanks a lot!

Perfect, that leaves only reading the button mapping from the mouse to be implemented. Could you capture the communication when opening RDCfg for the first time or connecting the mouse with RDCfg running for that?

>  sadly i haven't found a way to refresh RDCfg on Windows to see what dpi i just applied on linux

If i understand you correctly, you are setting the DPI on linux woth mouse_m908, then get the actual value with RDCfg on wondows. Doing the reverse (setting woth RDCfg, checking with mouse_m908) might be easier.

> would mouse_m908 work on Win10?

There is no obvious reason it wouldn't work since all dependencies (libusb, c++ compile, make or cmake) should be available on windows as well.

1nekomata, 2021-11-27:
> Perfect, that leaves only reading the button mapping from the mouse to be implemented. Could you capture the communication when opening RDCfg for the first time or connecting the mouse with RDCfg running for that?
yes, hang on a sec

> Perfect, that leaves only reading the button mapping from the mouse to be implemented. Could you capture the communication when opening RDCfg for the first time or connecting the mouse with RDCfg running for that?
no, i'm actually trying to have RDCfg detect what dpi has been set using mouse_m908 in order to check if those values are actually correct

1nekomata, 2021-11-27:
> There is no obvious reason it wouldn't work since all dependencies (libusb, c++ compile, make or cmake) should be available on windows as well.

ok

1nekomata, 2021-11-27:
Also i cant seem to be able to apply 
dpi1=0x0d000000
dpi2=0x0f000000 
dpi3=0x12000000
dpi4=0x14000000
dpi5=0x16000000
but i can apply
dpi1=0x0d00
dpi2=0x0f00 
dpi3=0x1200
dpi4=0x1400
dpi5=0x1600

[24 more replies]