dokutan/mouse_m908 #3: Error reading macros from M908

daddi79, 2020-09-07:
Hi,
    I'm an Arch user and I've a brand new Redragon M908. 
Reading mouse config with `mouse_m908 -R config_2.ini` generate the file correctly.
When I read file contents I see

> ;## macro1
> ;# unknown, please report as bug: 81 1 0
> ;# up   mouse_left

The second line should be simply a `down mouse_left`  command that I've previously sent to the device with `mouse_m908 -m ./macro_click.macro -n 1`.

This file contains only

> down    mouse_left
> up      mouse_left

There is something that I'm doing wrong or there is a bug to watch for?

Thanks in advance!

dokutan, 2020-09-07:
Thanks for opening this issue, this was indeed a bug in the macro decoding you found. I could reproduce this behaviour, the most recent commit i made should fix it (it did for my at least). Feel free to close this issue if it works for you, otherwise i will continue looking into this.

PS: i don't know what setup you are trying to achieve, but if you want to map a button as a left click you can simply use
```
button_12=left
```
in your configuration (obviously replace button_12 with the right one).

daddi79, 2020-09-08:
Thanks for your response! 
With the latest commit everything works fine for me, the issue can be closed :)

I'm trying to create a more complex macro, the example that I've provided was the simpliest where the issue could be found.

Thank you!

dokutan, 2020-09-08:
Perfect, then i am closing this.

closed 2020-09-08 by dokutan