dokutan/mouse_m908 #4: Figuring out DPI values.

SufoJareth, 2020-09-20:
Very new to this kind of thing but have gotten it completely set up for my M908. Was wondering how you figure out exact DPI values to put into the mouse config file.

`# dpi (4-8c)
dpi1=04
dpi2=16
dpi3=2d
dpi4=43
dpi5=8c`

Those are the examples listed and was wondering exactly how you would determine the value needed for an exact DPI number.

dokutan, 2020-09-20:
The values from the example are just the default values. To get from a real world DPI value (e.g. 12000) to the correct number (e.g. 8a, idk if these examples match) you would need a formula or lookup table. I have previously looked into this, but haven't found a solution.
What i've been doing so far is to just guess the numbers (i don't need exact DPI values anyway, lower number = lower DPI, higher number = higher DPI).

Hope this helps at least a bit, but i will look into this issue again.

dokutan, 2020-09-24:
If you are still interested, the 3.0 release allows you to specify the actual DPI value in the configuration file. Feel to open a new issue if you encounter any problems with the updated version.

closed 2020-09-24 by dokutan