dokutan/mouse_m908 #5: M990 Legend (not RGB/Chroma) support

rfxDarth, 2020-11-12:
Hey there. Been watching this project for a long time. Keep up the great work!

Unfortunately, though, the current build does not support my M990 Legend(PN70389) with any of the available profiles. And I've been too busy to capture the USB data... until now. 

So, here it goes: https://github.com/rfxDarth/m990_capture_data

If there's anything else I can provide or help you with, please, let me know.

Note, though, that this is different from M990 Legend Chroma(PN 78345) as the software is incompatible. There might have been even been multiple revisions, as the software they offer for 70389 is broken too. So, for anyone who might stumble here from google, I've uploaded an older but working version to https://github.com/rfxDarth/m990_legend-software.

dokutan, 2020-11-12:
Hello and thanks, i just saw this issue and had a brief look at the captures. From the official software, that you posted, the USB PID appears to be 0xfc0f, and therefore different to the  other M990. I have added the PID to the list of known PIDs, this should hopefully allow you to open the mouse and maybe change the profile with the existing generic support. If that doesn't work, could you post the output from ``lsusb`` (or an equivalent)?

There are a few questions that i have now:
- What buttons (including scrollwheel) does the mouse have that can be remapped?
- You have captures regarding LEDs, does the official software include LED controls, but the hardware does not have LEDs or what do you mean by "not RGB"?
- DPI minimum is 50, maximum is 16400?

dokutan, 2020-11-12:
I have added a new branch ``m990`` with a M990-specific backend. As it turns out the data layout is different for the M990, this also explains the broken official software.

Currently implemented features (i will update this over time), any feedback is welcome:
- Changing the profile
- Changing the settings to the default settings (call with ``-c some_random_file``)

rfxDarth, 2020-11-12:
In my previous experiments, running `sudo ./mouse_m908 -b 1 -d 2 -M generic -p 2` did nothing, except for sometimes causing mouse to become unresponsive until reconnected. But it did so with my own experiments at changing LED color too.

lsusb -vv output:
`Bus 001 Device 002: ID 04d9:fc0f Holtek Semiconductor, Inc. USB Gaming Mouse
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x04d9 Holtek Semiconductor, Inc.
  idProduct          0xfc0f 
  bcdDevice            2.00
  iManufacturer           0 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      47
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     106
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              21

`

- What buttons (including scrollwheel) does the mouse have that can be remapped?

All 24 of them, except that there must be at least one left mouse click button somewhere.

- You have captures regarding LEDs, does the official software include LED controls, but the hardware does not have LEDs or what do you mean by "not RGB"?

Yes, the mouse includes a programmable LED. What I mean by "not RGB" is that there are two distinct models, **M990 Legend** and **M990 Legend Chroma**, that seems to be called **M990 Legend RGB** in US market. Here they are in RU site

https://redragon.ru/catalog/gaming-mice/legend-laser-24-buttons-16400-dpi

https://redragon.ru/catalog/gaming-mice/legend-chroma-rgb-24000dpi-24-buttons

I'm not quite sure what's the difference between them, but Chroma seems to have more LED's in it and also more lighting options. Here are the differences in GUI options side by side.

[image]

- DPI minimum is 50, maximum is 16400?

Yes

- Edit: there are 5 different profiles?

Also yes. Although, the software I'm using is somewhat broken. The UI implies that polling rate, pointer speed, acceleration and color schemes are different for each profile, but only the currently selected profile's settings are applied and do they do change when profile is switched.

dokutan, 2020-11-12:
Thanks for the info, that really helped to get a general understanding of the features.

Unfortunately i am now certain that this mouse uses a different data layout (not such a big deal) and different keycodes, led mode codes, etc. (bigger problem) than the other known mice. This makes it basically impossible to reverse engineer without capturing an amount of data and immediate experimentation that is only practical with access to the hardware.

Therefore the situation is as follows:
- As described above, some features are implemented in the ``m908`` branch. This would be the starting point to continue work on this issue.
- You said that you did your own experiments, however i do not know how far you've come. To implement changing the settings the commented out sections of the ``write_settings()`` function in ``include/m990/writers.cpp`` need to be rewritten.

I am sorry that i could not be of much help to you. Your contributions are welcome, and i will support you as best as i can if you decide to start working on this.

Zeroamer, 2021-05-01:
> So, for anyone who might stumble here from google, I've uploaded an older but working version to https://github.com/rfxDarth/m990_legend-software.

Thank you so much! You are a life saver! I've been looking for this software for ages and I decide to look at this repository (and it's issues) one more time in a false hope that compatibility for this mouse would be added. Sadly, compatibility has not been added, but you are genuinely a life saver for uploading your copy of the software. Thank you again!