dokutan/mouse_m908 #6: Safety?

aleksejrs, 2020-11-14:
How likely is it that a mouse would be bricked due to its own bug, or a bug in the official software (does the latter work on WINE?)? I found a review of M990 Legend Chroma RGB, where the author says two mice bricked after a week, which he thought was from frequent reprogramming with the official software. With Firestorm (IIRC), the Reset button would help, but M990 has no such button. He bought another one and didn't reprogram it, and it didn't brick.

dokutan, 2020-11-14:
> How likely is it that a mouse would be bricked due to its own bug, or a bug in the official software ?

That is something i can only estimate, i had almost no problems with the M908, i only managed to brick it by disabling all DPI levels, however this could be fixed with the official software and is now prevented by this software. I would assume that there has to be a bug in the firmware of the mouse for it to be bricked by the official software.

> does the latter work on WINE?

I couldn't get the official software to work in WINE (and i think it is unlikely to be possible due to the required hardware access).

> frequent reprogramming with the official software.

The settings on the mouse are most likely stored in flash memory, in theory it could be possible to wear that out (https://en.wikipedia.org/wiki/Flash_memory#Memory_wear). Or there is a bug in the official software/firmware.

If you have a M990 Chroma, you should however be aware of the disclaimer in the README and that this software is untested.
I hope this answered your questions and would be glad to hear about your results should you try out this software.

aleksejrs, 2020-11-14:
I asked about WINE because of this comment about a graphics tablet on AliExpress: https://github.com/DIGImend/digimend-kernel-drivers/issues/182#issuecomment-480510535
What might clarify that comment:
https://github.com/DIGImend/digimend-kernel-drivers/issues/182#issuecomment-489413547
https://github.com/DIGImend/digimend-kernel-drivers/issues/182#issuecomment-624120315

I ordered the mouse, but it probably wasn't an official store, but some other store on AliExpress with similar ratings for the store and the mouse (it was cheaper; that store still has M908 more expensive than M990 currently is in the official store). So it's probably without warrantly, and it might have made much more sense to buy it at a similar price in Russia, where the company appears to be, or the M908 if I had noticed the driver.

Assuming it arrives (next month), the default setting will still be useful. Can I help without Windows and risking? Does it have multiple profiles by default?

dokutan, 2020-11-14:
> Can I help without Windows and risking?

You can try to change the profile with this software, the risk should be minimal, but technically not zero. To fully add support to this software you most likely would have to use Windows (inside a VM or on real hardware) to run the official software and capture the usb communication with wireshark.

> Does it have multiple profiles by default?

I know nothing about the M990 Chroma, but all other mice that i am aware of have 5 profiles by default, and this number can not be changed.

> Does it need a mousepad, compared to a wooden table (≈particle board)?

I use my M908 on a wooden table without any problems, however your mouse might have a different sensor.

dokutan, 2021-01-04:
I'm closing this issue as all questions seem to be answered. If you, the author or anyone else who is reading this, have any other question, feel free to ask in a new issue.

closed 2021-01-04 by dokutan