dokutan/mouse_m908 #7: M990 Legend Chroma settings read after pressing buttons

aleksejrs, 2020-12-20:
Here are the settings read after pressing the DPI+, LED and profile buttons.

When switching profile, the `Currently active profile` changes.
When switching DPI, `Active dpi level for this profile:` changes.
When switching LED mode, lightmode and sometimes speed change.

Profile switching also works.

dokutan, 2020-12-20:
Thanks for the info, i have added experimental support for the M990 Chroma in the master branch and updated the README. Could you attach the output of ``mouse_m908 -D -`` and ``mouse_m908 -R -`` after building from the master branch?

If you want to improve the functionality for this mouse, it would be best to set up the official software and capture the USB communication with wireshark when changing the settings. Otherwise you could try to change the settings with this software and see what works.

aleksejrs, 2020-12-20:
> Could you attach the output of `mouse_m908 -D -` and `mouse_m908 -R -` after building from the master branch?
[mouse_m908-dump-2.txt](https://github.com/dokutan/mouse_m908/files/5721566/mouse_m908-dump-2.txt)
[mouse_m908-read-2.txt](https://github.com/dokutan/mouse_m908/files/5721567/mouse_m908-read-2.txt)

dokutan, 2020-12-22:
That was quite helpful, reading the button mapping should now work correctly for the first 20 buttons. Implementing any other features would require capturing the USB communication from the official software.

dokutan, 2021-04-05:
I'm closing this due to inactivity and because there doesn't seem to be anything to add about reading the settings on this mouse.
For anyone reading this, an issue concerning proper support is #18, or open a new issue if there is anything else.

closed 2021-04-05 by dokutan