dokutan/mouse_m908 #10: M801 Mammoth compatibility

Metaph413, 2021-01-10:
The M801 Mammoth is my main mouse. What can I do to help it be compatible with this software?

dokutan, 2021-01-10:
The first step would be to run ``lsusb``, ``mouse_m908 -D -`` and ``mouse_m908 -R -`` and post the output. That would allow the creation of an experimental backend where features can be added or implemented. The second step would then be for you to test the features as they are implemented, and capture the USB traffic of the official software with wireshark if features are not working.
I hope this answers your question, feel free to ask if anything is unclear.

Metaph413, 2021-01-10:
I cloned the repository, but I have no idea what to do from there.

dokutan, 2021-01-10:
Build and install as described in the README, then run ``lsusb``, ``mouse_m908 -R -`` and ``mouse_m908 -D -`` and attach the output to your comment. If you have problems with compiling, the basic steps are:
1. install the dependencies: libusb (dev package on Debian based distros), make, pkgconf, gcc/g++
2. ``cd /path/to/mouse_m908``
3. ``make``
4. ``sudo make install``

Metaph413, 2021-01-17:
When I run "mouse_m908 -R -" and "mouse_m908 -D -", I get a Couldn't detect mouse error message. What am I missing?

dokutan, 2021-01-17:
You could run ``lsusb`` to check if the USB IDs are different from 04d9:fc56, then use the ``--bus`` and ``--device`` options, or try running the program as root. If you post the output of ``lsusb`` i might be able to give you more information.

Metaph413, 2021-01-17:
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0cf3:e009 Qualcomm Atheros Communications 
Bus 001 Device 007: ID 04d9:fa56 Holtek Semiconductor, Inc. USB Laser Game Mouse
Bus 001 Device 005: ID 413c:2011 Dell Computer Corp. Multimedia Pro Keyboard
Bus 001 Device 003: ID 413c:1005 Dell Computer Corp. Multimedia Pro Keyboard Hub
Bus 001 Device 002: ID 1038:12ad SteelSeries ApS SteelSeries Arctis 7
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I believe my mouse is 7.

dokutan, 2021-01-17:
Thanks that helped, the USB PID was indeed wrong. I have fixed it in the master branch.

Metaph413, 2021-01-18:
How do I update it? And what commands do I run afterwards to get the mouse info you need?

dokutan, 2021-01-18:
Assuming that you still have the repositrory cloned, if not clone this repository first (``git clone https://github.com/dokutan/mouse_m908``):
```
cd path/to/mouse_m908
git pull
make clean
make
sudo make install
```

Metaph413, 2021-01-19:
OK, it seems to have worked. Here are the outputs.
[D Output.txt](https://github.com/dokutan/mouse_m908/files/5838820/D.Output.txt)
[R Output.txt](https://github.com/dokutan/mouse_m908/files/5838821/R.Output.txt)

dokutan, 2021-01-19:
I'm glad that worked, unfortunately from what i can tell now is this mouse is using a different data format, maybe similar to the M990 (#5). This means it would be neccessary to install the official software in a VM with USB passthrough or on bare metal and use wireshark to capture the USB data when changing settings.

In case that is something you want to try, here are a few links to explain the process (you don't have to do the reverse engineering part, only the capture), or feel free to ask here.
https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc (from my other usb reverse engineering project)

I'm sorry i couldn't be of much help, but please let me know if you want to continue to work on this or not.