dokutan/mouse_m908 #14: PKGBUILD for Arch Linux

dokutan, 2021-02-10:
Current situation: At least three people (@EsdrasTarsis, @nwildner, @dokutan) are interested in a PKGBUILD for Arch Linux. At first @nwildner created a fork two days ago, that is now gone with intent to continue work on it (https://github.com/dokutan/mouse_m908/issues/2#issuecomment-776345330). Now @EsdrasTarsis has opened a PR (#13). I am creating this issue because i don't know what the best action would be and to provide a place for further discussion.

Therefore i am asking you, @EsdrasTarsis, @nwildner what are your thoughts on this matter?

nwildner, 2021-02-10:
Well, i know @EsdrasTarsis from outside github, and he got this pkg build working first(he even sent me his package through pm on Telegram)

You can merge his PKGBUILD and i will help him if needed :)

dokutan, 2021-02-10:
Pefect, thanks for the info. I have merged it now and am closing this issue as well, as the better place for discussion of further changes are the appropriate PRs.

closed 2021-02-10 by dokutan