dokutan/mouse_m908 #15: Would it be possible to add support for M913 (wireless model) ?

anvena, 2021-02-12:


dokutan, 2021-02-13:
Hard to tell without any information, the description on the website claims that you can connect the mouse via a wireless dongle or a  USB cable. Could you please post the output of ``lsusb`` when using the wireless dongle and the cable?

anvena, 2021-02-14:
Many thanks for response. So output of lsusb is:
`lsusb
Bus 002 Device 003: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 009: ID 2109:0813 VIA Labs, Inc. VL813 Hub
Bus 002 Device 008: ID 2109:0813 VIA Labs, Inc. VL813 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0b05:1914 ASUSTek Computer, Inc. ROG Gaming Display Aura Device
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 034: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 036: ID 1532:0235 Razer USA, Ltd BlackWidow Lite (2018)
Bus 001 Device 039: ID 25a7:fa07 Areson Technology Corp 2.4G Wireless Receiver - wireless option
Bus 001 Device 040: ID 25a7:fa08 Areson Technology Corp 2.4G Dual Mode Mouse - usb cable
Bus 001 Device 031: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 001 Device 030: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 001 Device 009: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
And I tried to execute commands you provided but no luck:

`
[mouse_m908]$ mouse_m908 -R -
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.

[mouse_m908]$ mouse_m908 -D -
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.

mouse_m908]$ mouse_m908 -p 1
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.

`

dokutan, 2021-02-14:
The USB ids are indeed correct, just to rule out any weird bugs, have you tried to run mouse_m908 with root permissions (i.e. ``sudo mouse_m908 -R -``)?

anvena, 2021-02-15:
I tried options below:
`
[manjaro ~]$ lsusb
Bus 002 Device 010: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 009: ID 2109:0813 VIA Labs, Inc. VL813 Hub
Bus 002 Device 008: ID 2109:0813 VIA Labs, Inc. VL813 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 045: ID 0b05:1914 ASUSTek Computer, Inc. ROG Gaming Display Aura Device
Bus 001 Device 044: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 034: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 036: ID 1532:0235 Razer USA, Ltd BlackWidow Lite (2018)
Bus 001 Device 041: ID 25a7:fa07 Areson Technology Corp 2.4G Wireless Receiver
Bus 001 Device 031: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 001 Device 030: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 001 Device 009: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[manjaro ~]$ sudo mouse_m908 -R -
[sudo] password for tol: 
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.

[manjaro ~]$ sudo mouse_m908 -R --bus 001 --device 041
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.

[manjaro ~]$ sudo mouse_m908 -p 1 --bus 001 --device 25a7:fa07
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.

`

dokutan, 2021-02-15:
That means the M913 uses a different protocol with at least different USB endpoints. To implement support it would be neccessary to capture the USB communication from the official software when changing the settings with wireshark, here are a few links that should explain the process, or ask here if you have any questions.

https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc (from my other usb reverse engineering project)

If you are not interested or not able to capturr USB data, that is no problem for me but please tell me clearly 
so this issu

anvena, 2021-02-15:
Many thanks for response, unfortunately I'm not expert in this area so would not be able to implement it by myself but if you could guide me I could provide information which might be used to fix issue with M913.

dokutan, 2021-02-15:
The first step would be to install the official software on Windows either in a virtual machine with USB passthrough or on bare metal. Then install wireshark (on the host if you use a VM) and capture the communication with the mouse when changing specific settings on the mouse, i.e. start a capture, change some setting, save the capture, repeat.

Basically just follow the directions in the first link i posted above. I hope this helps enough to get you started.

anvena, 2021-02-20:
So I have win10 installed on VM should I install Wireshark on manjaro host or on windows (which is on KVM) ?

dokutan, 2021-02-20:
Both ways should work, i personally have only used it on the Linux host and would recommend that.

anvena, 2021-02-24:
Thank you for response. I tried to install Wireshark but did not find how to  capture usb port:
[image]

dokutan, 2021-02-24:
I am assuming you are on Linux, have you loaded the usbmon kernel module with ``sudo modprobe usbmon`` ?

anvena, 2021-02-26:
OK so seems like I got it how to start and apply filter :). If I followed nest steps like let's  say go to settings assign new key for each button and for each change would save captured report, would it be enough ?

dokutan, 2021-02-26:
The idea is to do these steps for any setting that you might want to implement, i.e.
- changing all physical buttons individually
- changing and enabling/disabling all DPI levels individually
- LED brightness, color, mode, speed
- USB poll rate
- all other interesting available settings

It is hard to tell what might be enough without any understanding of the protocol. The more the better, the same applies for documenting what you changed.

anvena, 2021-02-26:
I've put all files into repo anvena/redragon-M913. Could you please check if it's enough ?

dokutan, 2021-02-26:
I'm sorry but i can't find your repo, have you maybe set it to private or am i missing something else?

anvena, 2021-02-27:
Yes it was private and you had an access but nevertheless I made it public to simplify access

dokutan, 2021-02-28:
Thanks, that was really helpful. The data you captured confirms that the mouse is indeed using a completely different protocol but support should be possible. I have created a new [branch](https://github.com/dokutan/mouse_m908/tree/m913) with a backend for the M913. Currently the goal is only to be able to talk to the mouse, if that works i will be implementing the actual features.

For now could you please try building from that branch and running ``mouse_m908 -R -`` to test if it is possible to open the mouse, as i had to do some guessing about the USB interfaces. If that works (it should output ``Not implemented, missing data``) you can try to run ``mouse_m908 -c some_empty_file``, to test if sending data works. If opening the mouse fails, could you please add the output of ``lsusb -v -d 25a7:fa07`` and ``lsusb -v -d 25a7:fa08``.

I noticed that there was no data for changing the active profile, does the mouse not have profiles or is
 it not possible to change them through the software? If the answers are no and no, could you capture data for that and tell me how many profiles there are?

anvena, 2021-03-01:
Hi, I've just tried your branch, please check result:
```
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R -
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R - lsusb -v -d 25a7:fa07        
Version: 3.1
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.
[tol@tol-manjaro mouse_m908]$ lsusb -v -d 25a7:fa07

Bus 001 Device 003: ID 25a7:fa07 Areson Technology Corp 2.4G Wireless Receiver
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x25a7 Areson Technology Corp
  idProduct          0xfa07 
  bcdDevice           13.00
  iManufacturer           1 Compx
  iProduct                2 2.4G Wireless Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      87
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0007  1x 7 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     189
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0011  1x 17 bytes
        bInterval               1
[tol@tol-manjaro mouse_m908]$ lsusb -v -d 25a7:fa08
[tol@tol-manjaro mouse_m908]$ git status
On branch m913
Your branch is up to date with 'origin/m913'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        mouse_m908-3.1-1-x86_64.pkg.tar.zst
        src/
        v3.1.tar.gz

nothing added to commit but untracked files present (use "git add" to track)

```

anvena, 2021-03-01:
I'll try to check profiles for mouse, I saw options "Import Profile" and "Export Profile". Will try to play with profiles.

dokutan, 2021-03-01:
Thanks, the lsusb output looks good. As for the failure to detect the mouse, i honestly have no idea what is going on exactly, as it works for me when i change the IDs on the M913 backend to the IDs of my mouse. I have added some debug output for the detection on the m913 branch, could you please try it with the following commands:
```
sudo mouse_m908 -R -
sudo mouse_m908 -R - -M913
sudo mouse_m908 -R - --bus 1 --device 3 # look up the correct device id by running lsusb
sudo mouse_m908 -R - --bus 1 --device 3 -M913
```

anvena, 2021-03-01:
I've just captured data for profiles and updated my repository so new files have been uploaded.
There are 2 in memory profiles which might be changed by pressing button on the mouse. 

 - mode_1.pcapng and mode_2.pcapng files keep captured information when switch between them.  
 - mode1.jmk and mode2.jmk exported files for each in memory profile

anvena, 2021-03-01:
Update: Mouse does not work without wifi receiver that is when I used just wired connection. It seems that usb cable is used just for charging.

dokutan, 2021-03-01:
If i understand you correctly it doesn't work with either the wired or wireless connection. Could you make certain that you are using a build that includes a04dbcf70552d98e836b19ca2b63fff7cb38ff23, maybe call the binary from directly from the repo with ``sudo ./mouse_m908 -R -``, it should print a list of all usb devices that are connected during the mouse detection. If it doesn't the problem seems to be with libusb.

anvena, 2021-03-01:
My log:

`
[tol@tol-manjaro mouse_m908]$ git log  
commit a04dbcf70552d98e836b19ca2b63fff7cb38ff23 (HEAD -> m913, origin/m913)
Author: dokutan <54861821+dokutan@users.noreply.github.com>
Date:   Mon Mar 1 21:38:55 2021 +0100

`

then build package:

`
tol@tol-manjaro mouse_m908]$ makepkg -si
==> WARNING: A package has already been built, installing existing package...
==> Installing package mouse_m908 with pacman -U...
[sudo] password for tol: 
loading packages...
warning: mouse_m908-3.1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) mouse_m908-3.1-1

Total Installed Size:  0.55 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                                                                  [##################################################################] 100%
(1/1) checking package integrity                                                                                [##################################################################] 100%
(1/1) loading package files                                                                                     [##################################################################] 100%
(1/1) checking for file conflicts                                                                               [##################################################################] 100%
(1/1) checking available disk space                                                                             [##################################################################] 100%
:: Processing package changes...
(1/1) reinstalling mouse_m908                                                                                   [##################################################################] 100%
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...

`

And finally commands:

`
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R -
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.
[tol@tol-manjaro mouse_m908]$ sudo ./mouse_m908 -R -
sudo: ./mouse_m908: command not found
[tol@tol-manjaro mouse_m908]$ cd ..
[tol@tol-manjaro ~]$ sudo ./mouse_m908 -R -
sudo: ./mouse_m908: command not found
[tol@tol-manjaro ~]$ cd mouse_m908/
[tol@tol-manjaro mouse_m908]$ sudo ./mouse_m908 -R -
sudo: ./mouse_m908: command not found
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R -
Couldn't detect mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with the --model option
If nothing works please report this as a bug.
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R - -M913
Unknown model, valid options are:
709
711
715
908
generic

`

dokutan, 2021-03-01:
That explains everything, the PKGPUILD downloads and installs the release version. You need to run ``make && sudo make install`` to get the git version, or don't install and run from the local directory.
I'm really sorry for that misunderstanding, i will add a note in the README making this more clear.

anvena, 2021-03-01:
```
[tol@tol-manjaro mouse_m908]$ make && sudo make install
c++ -c include/m908/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m908.o
c++ -c include/m908/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m908.o
c++ -c include/m908/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m908.o
c++ -c include/m908/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m908.o
c++ -c include/m908/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m908.o
c++ -c include/m908/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m908.o
c++ -c include/m908/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m908.o
c++ -c include/m709/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m709.o
c++ -c include/m709/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m709.o
c++ -c include/m709/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m709.o
c++ -c include/m709/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m709.o
c++ -c include/m709/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m709.o
c++ -c include/m709/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m709.o
c++ -c include/m709/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m709.o
c++ -c include/m711/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m711.o
c++ -c include/m711/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m711.o
c++ -c include/m711/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m711.o
c++ -c include/m711/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m711.o
c++ -c include/m711/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m711.o
c++ -c include/m711/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m711.o
c++ -c include/m711/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m711.o
c++ -c include/m715/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m715.o
c++ -c include/m715/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m715.o
c++ -c include/m715/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m715.o
c++ -c include/m715/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m715.o
c++ -c include/m715/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m715.o
c++ -c include/m715/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m715.o
c++ -c include/m715/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m715.o
c++ -c include/m913/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m913.o
c++ -c include/m913/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m913.o
c++ -c include/m913/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m913.o
c++ -c include/m913/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m913.o
c++ -c include/m913/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m913.o
c++ -c include/m913/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m913.o
c++ -c include/m913/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m913.o
c++ -c include/m990/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m990.o
c++ -c include/m990/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m990.o
c++ -c include/m990/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m990.o
c++ -c include/m990/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m990.o
c++ -c include/m990/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m990.o
c++ -c include/m990/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m990.o
c++ -c include/m990/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m990.o
c++ -c include/m990chroma/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_m990chroma.o
c++ -c include/m990chroma/data.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o data_m990chroma.o
c++ -c include/m990chroma/getters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o getters_m990chroma.o
c++ -c include/m990chroma/helpers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o helpers_m990chroma.o
c++ -c include/m990chroma/setters.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o setters_m990chroma.o
[... cut ...]
c++ *.o -o mouse_m908 -lusb-1.0  -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0`
[sudo] password for tol: 
cp ./mouse_m908 /usr/bin/mouse_m908 && \
cp ./mouse_m908.rules /etc/udev/rules.d && \
mkdir /usr/share/doc/mouse_m908 | true && \
cp ./examples/* /usr/share/doc/mouse_m908/ && \
cp ./README.md /usr/share/doc/mouse_m908/ && \
cp ./keymap.md /usr/share/doc/mouse_m908/ && \
cp ./mouse_m908.1 /usr/share/man/man1/
mkdir: cannot create directory ‘/usr/share/doc/mouse_m908’: File exists
[tol@tol-manjaro mouse_m908]$ 

```

anvena, 2021-03-01:
```
[tol@tol-manjaro mouse_m908]$ mouse_m908 -R -
generic 1d6b:3  false
990chroma       1d6b:3  false
990     1d6b:3  false
913     1d6b:3  false
908     1d6b:3  false
715     1d6b:3  false
711     1d6b:3  false
709     1d6b:3  false
        1d6b:3  false
generic 25a7:fa07       false
990chroma       25a7:fa07       false
990     25a7:fa07       false
913     25a7:fa07       true
908     25a7:fa07       false
715     25a7:fa07       false
711     25a7:fa07       false
709     25a7:fa07       false
        25a7:fa07       false
generic 46d:825 false
990chroma       46d:825 false
990     46d:825 false
913     46d:825 false
908     46d:825 false
715     46d:825 false
711     46d:825 false
709     46d:825 false
        46d:825 false
generic 1532:235        false
990chroma       1532:235        false
990     1532:235        false
913     1532:235        false
908     1532:235        false
715     1532:235        false
711     1532:235        false
709     1532:235        false
        1532:235        false
generic 1005:b113       false
990chroma       1005:b113       false
990     1005:b113       false
913     1005:b113       false
908     1005:b113       false
715     1005:b113       false
711     1005:b113       false
709     1005:b113       false
        1005:b113       false
generic 8564:1000       false
990chroma       8564:1000       false
990     8564:1000       false
913     8564:1000       false
908     8564:1000       false
715     8564:1000       false
711     8564:1000       false
709     8564:1000       false
        8564:1000       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 8087:aaa        false
990chroma       8087:aaa        false
990     8087:aaa        false
913     8087:aaa        false
908     8087:aaa        false
715     8087:aaa        false
711     8087:aaa        false
709     8087:aaa        false
        8087:aaa        false
generic 1d6b:2  false
990chroma       1d6b:2  false
990     1d6b:2  false
913     1d6b:2  false
908     1d6b:2  false
715     1d6b:2  false
711     1d6b:2  false
709     1d6b:2  false
        1d6b:2  false
Couldn't open mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with or without the --kernel-driver option
- Try with the --model option
- Try with the --bus and --device options
If nothing works please report this as a bug.
[tol@tol-manjaro mouse_m908]$ 

```

anvena, 2021-03-01:
```
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R - -M913
generic 1d6b:3  false
990chroma       1d6b:3  false
990     1d6b:3  false
913     1d6b:3  false
908     1d6b:3  false
715     1d6b:3  false
711     1d6b:3  false
709     1d6b:3  false
        1d6b:3  false
generic 25a7:fa07       false
990chroma       25a7:fa07       false
990     25a7:fa07       false
913     25a7:fa07       true
908     25a7:fa07       false
715     25a7:fa07       false
711     25a7:fa07       false
709     25a7:fa07       false
        25a7:fa07       false
generic 46d:825 false
990chroma       46d:825 false
990     46d:825 false
913     46d:825 false
908     46d:825 false
715     46d:825 false
711     46d:825 false
709     46d:825 false
        46d:825 false
generic 1532:235        false
990chroma       1532:235        false
990     1532:235        false
913     1532:235        false
908     1532:235        false
715     1532:235        false
711     1532:235        false
709     1532:235        false
        1532:235        false
generic 1005:b113       false
990chroma       1005:b113       false
990     1005:b113       false
913     1005:b113       false
908     1005:b113       false
715     1005:b113       false
711     1005:b113       false
709     1005:b113       false
        1005:b113       false
generic 8564:1000       false
990chroma       8564:1000       false
990     8564:1000       false
913     8564:1000       false
908     8564:1000       false
715     8564:1000       false
711     8564:1000       false
709     8564:1000       false
        8564:1000       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 8087:aaa        false
990chroma       8087:aaa        false
990     8087:aaa        false
913     8087:aaa        false
908     8087:aaa        false
715     8087:aaa        false
711     8087:aaa        false
709     8087:aaa        false
        8087:aaa        false
generic 1d6b:2  false
990chroma       1d6b:2  false
990     1d6b:2  false
913     1d6b:2  false
908     1d6b:2  false
715     1d6b:2  false
711     1d6b:2  false
709     1d6b:2  false
        1d6b:2  false
# Model: 913
Not implemented, missing data
[tol@tol-manjaro mouse_m908]$ 

```

anvena, 2021-03-01:
```
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -R - --bus 1 --device 3
generic 1d6b:3  false
990chroma       1d6b:3  false
990     1d6b:3  false
913     1d6b:3  false
908     1d6b:3  false
715     1d6b:3  false
711     1d6b:3  false
709     1d6b:3  false
        1d6b:3  false
generic 25a7:fa07       false
990chroma       25a7:fa07       false
990     25a7:fa07       false
913     25a7:fa07       true
908     25a7:fa07       false
715     25a7:fa07       false
711     25a7:fa07       false
709     25a7:fa07       false
        25a7:fa07       false
generic 46d:825 false
990chroma       46d:825 false
990     46d:825 false
913     46d:825 false
908     46d:825 false
715     46d:825 false
711     46d:825 false
709     46d:825 false
        46d:825 false
generic 1532:235        false
990chroma       1532:235        false
990     1532:235        false
913     1532:235        false
908     1532:235        false
715     1532:235        false
711     1532:235        false
709     1532:235        false
        1532:235        false
generic 1005:b113       false
990chroma       1005:b113       false
990     1005:b113       false
913     1005:b113       false
908     1005:b113       false
715     1005:b113       false
711     1005:b113       false
709     1005:b113       false
        1005:b113       false
generic 8564:1000       false
990chroma       8564:1000       false
990     8564:1000       false
913     8564:1000       false
908     8564:1000       false
715     8564:1000       false
711     8564:1000       false
709     8564:1000       false
        8564:1000       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 8087:aaa        false
990chroma       8087:aaa        false
990     8087:aaa        false
913     8087:aaa        false
908     8087:aaa        false
715     8087:aaa        false
711     8087:aaa        false
709     8087:aaa        false
        8087:aaa        false
generic 1d6b:2  false
990chroma       1d6b:2  false
990     1d6b:2  false
913     1d6b:2  false
908     1d6b:2  false
715     1d6b:2  false
711     1d6b:2  false
709     1d6b:2  false
        1d6b:2  false
Segmentation fault
[tol@tol-manjaro mouse_m908]$ 

```

dokutan, 2021-03-01:
Perfect, that means the mouse gets detected and opened. You can try ``sudo mouse_m908 -c some_empty_file`` to try sending some of the captured data.

anvena, 2021-03-01:
Seems like after uploading file my active profile has been changed:

```
[tol@tol-manjaro mouse_m908]$ sudo mouse_m908 -c empty
generic 1d6b:3  false
990chroma       1d6b:3  false
990     1d6b:3  false
913     1d6b:3  false
908     1d6b:3  false
715     1d6b:3  false
711     1d6b:3  false
709     1d6b:3  false
        1d6b:3  false
generic 25a7:fa07       false
990chroma       25a7:fa07       false
990     25a7:fa07       false
913     25a7:fa07       true
908     25a7:fa07       false
715     25a7:fa07       false
711     25a7:fa07       false
709     25a7:fa07       false
        25a7:fa07       false
generic 46d:825 false
990chroma       46d:825 false
990     46d:825 false
913     46d:825 false
908     46d:825 false
715     46d:825 false
711     46d:825 false
709     46d:825 false
        46d:825 false
generic 1532:235        false
990chroma       1532:235        false
990     1532:235        false
913     1532:235        false
908     1532:235        false
715     1532:235        false
711     1532:235        false
709     1532:235        false
        1532:235        false
generic 1005:b113       false
990chroma       1005:b113       false
990     1005:b113       false
913     1005:b113       false
908     1005:b113       false
715     1005:b113       false
711     1005:b113       false
709     1005:b113       false
        1005:b113       false
generic 8564:1000       false
990chroma       8564:1000       false
990     8564:1000       false
913     8564:1000       false
908     8564:1000       false
715     8564:1000       false
711     8564:1000       false
709     8564:1000       false
        8564:1000       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 8087:aaa        false
990chroma       8087:aaa        false
990     8087:aaa        false
913     8087:aaa        false
908     8087:aaa        false
715     8087:aaa        false
711     8087:aaa        false
709     8087:aaa        false
        8087:aaa        false
generic 1d6b:2  false
990chroma       1d6b:2  false
990     1d6b:2  false
913     1d6b:2  false
908     1d6b:2  false
715     1d6b:2  false
711     1d6b:2  false
709     1d6b:2  false
        1d6b:2  false

```

dokutan, 2021-03-01:
That is as expected, i took the relevant data from one of you captures, when changing the settings the idea is to modify this packet to include the changes, then send it to the mouse, however as no settings are currently implemented these default settings are applied. This means it is now possible to talk to the mouse and i can focus on implementing the settings.

anvena, 2021-03-01:
Great, thank you for your time. Going to disconnect so if you need anything else will check tomorrow.

dokutan, 2021-03-02:
I'm beginning to get an understanding of the protocol, here is a list of things that require capture:
- Changing the first side button and the normal mouse buttons to two different values
- Changing the first DPI level to something other than 1000 or 100
- Enabling and Disabling off DPI levels
- For one DPI level (doesn't matter which) changing it to every possible value. This is to allow actual DPI values in the config instead of raw bytes.
- Changing to each available lightmode, while not changing the color/speed/brightness if possible
- Only changing brightness/lightmode speed to every possible value, and 2 different colors
- Capture while opening the software after closing it completeley from the taskmanager

This is beacuse the workflow is to extract the relevant bytes from the capture and then compare different captures to narrow down the position and meaning for each setting.

anvena, 2021-03-02:
That strange but seems like no effect when I change values in examples/example_m913.ini I'd say the same when I applied empty file. Also I noticed that button 1 stopped working. 

Will try to capture  some data.

dokutan, 2021-03-03:
#### Changing DPI
I rewrote the DPI handling in b921bd30473aa1981ad79ac36d67449a771ed95d, can you change the DPI on the mouse with that commit with ``mouse_m908 -c examples/example_m913.ini`` after modifying examples/example_m913.ini? Please post the output if it does not work.

#### DPI values
Only a few DPI values are currently available, as the capture you provided last showed, the DPI bytecode is not linear, i.e. captured data for all values is required. If changing the DPI as above works, please capture data for all values, however it is enough to combine multiple values in one capture, e.g. 1000 on level 1, 1100 on level 2, …

#### Enabling and Disabling off DPI levels
> Not POSSIBLE there is no such option

I was thinking that the checkbox that is visible next to the DPI sliders in your screenshot would do that, now i am curious what it does instead?

anvena, 2021-03-04:
**Changing DPI**

Switched to commit 

<details>
<summary>mouse_m908 -c examples/example_m913.ini</summary>
<p>
.ini file was changed to other descending ordering:

```
# DPI levels
# valid are: 100,200,300,400,500,600,700,800,900,1000,8000,16000
dpi1=16000
dpi2=8000
dpi3=1000
dpi4=500
dpi5=100
```

```
[tol@tol-manjaro mouse_m908]$ mouse_m908 -c examples/example_m913.ini
generic 1d6b:3  false
990chroma       1d6b:3  false
990     1d6b:3  false
913     1d6b:3  false
908     1d6b:3  false
715     1d6b:3  false
711     1d6b:3  false
709     1d6b:3  false
        1d6b:3  false
generic 25a7:fa07       false
990chroma       25a7:fa07       false
990     25a7:fa07       false
913     25a7:fa07       true
908     25a7:fa07       false
715     25a7:fa07       false
711     25a7:fa07       false
709     25a7:fa07       false
        25a7:fa07       false
generic 46d:825 false
990chroma       46d:825 false
990     46d:825 false
913     46d:825 false
908     46d:825 false
715     46d:825 false
711     46d:825 false
709     46d:825 false
        46d:825 false
generic 1532:235        false
990chroma       1532:235        false
990     1532:235        false
913     1532:235        false
908     1532:235        false
715     1532:235        false
711     1532:235        false
709     1532:235        false
        1532:235        false
generic 1005:b113       false
990chroma       1005:b113       false
990     1005:b113       false
913     1005:b113       false
908     1005:b113       false
715     1005:b113       false
711     1005:b113       false
709     1005:b113       false
        1005:b113       false
generic 8564:1000       false
990chroma       8564:1000       false
990     8564:1000       false
913     8564:1000       false
908     8564:1000       false
715     8564:1000       false
711     8564:1000       false
709     8564:1000       false
        8564:1000       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 8087:aaa        false
990chroma       8087:aaa        false
990     8087:aaa        false
913     8087:aaa        false
908     8087:aaa        false
715     8087:aaa        false
711     8087:aaa        false
709     8087:aaa        false
        8087:aaa        false
generic 1d6b:2  false
990chroma       1d6b:2  false
990     1d6b:2  false
913     1d6b:2  false
908     1d6b:2  false
715     1d6b:2  false
711     1d6b:2  false
709     1d6b:2  false
        1d6b:2  false
Warning: Unknown DPI value 16000
Warning: Unknown DPI value 8000
Warning: Unknown DPI value 1000
Warning: Unknown DPI value 500
Warning: Unknown DPI value 100
[tol@tol-manjaro mouse_m908]$ 

```

</details>

Unfortunately no effect after applying updated settings.

anvena, 2021-03-04:
**Changing DPI**

**DPI values**

**Enabling and Disabling off DPI levels**

**Button mapping**

Switched to commit 8143868 

<details>
<summary>mouse_m908 -c examples/example_m913.ini</summary>
<p>

.ini file:

```
# DPI levels
# valid are: 100,200,300,400,500,600,700,800,900,1000,8000,16000
dpi1=100
dpi2=200
dpi3=300
dpi4=8000
dpi5=16000

# Button mapping
# currently supported: forward, left, led_toggle, none
button_1=left
button_2=forward
button_3=led_toggle
button_4=none
button_5=none
button_6=none
button_7=none
button_8=none
button_9=none
button_10=none
button_11=none
button_12=none

```

```
[tol@tol-manjaro mouse_m908]$ mouse_m908 -c examples/example_m913.ini
generic 1d6b:3  false
990chroma       1d6b:3  false
990     1d6b:3  false
913     1d6b:3  false
908     1d6b:3  false
715     1d6b:3  false
711     1d6b:3  false
709     1d6b:3  false
        1d6b:3  false
generic 25a7:fa07       false
990chroma       25a7:fa07       false
990     25a7:fa07       false
913     25a7:fa07       true
908     25a7:fa07       false
715     25a7:fa07       false
711     25a7:fa07       false
709     25a7:fa07       false
        25a7:fa07       false
generic 46d:825 false
990chroma       46d:825 false
990     46d:825 false
913     46d:825 false
908     46d:825 false
715     46d:825 false
711     46d:825 false
709     46d:825 false
        46d:825 false
generic 1532:235        false
990chroma       1532:235        false
990     1532:235        false
913     1532:235        false
908     1532:235        false
715     1532:235        false
711     1532:235        false
709     1532:235        false
        1532:235        false
generic 1005:b113       false
990chroma       1005:b113       false
990     1005:b113       false
913     1005:b113       false
908     1005:b113       false
715     1005:b113       false
711     1005:b113       false
709     1005:b113       false
        1005:b113       false
generic 8564:1000       false
990chroma       8564:1000       false
990     8564:1000       false
913     8564:1000       false
908     8564:1000       false
715     8564:1000       false
711     8564:1000       false
709     8564:1000       false
        8564:1000       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 2109:2813       false
990chroma       2109:2813       false
990     2109:2813       false
913     2109:2813       false
908     2109:2813       false
715     2109:2813       false
711     2109:2813       false
709     2109:2813       false
        2109:2813       false
generic 8087:aaa        false
990chroma       8087:aaa        false
990     8087:aaa        false
913     8087:aaa        false
908     8087:aaa        false
715     8087:aaa        false
711     8087:aaa        false
709     8087:aaa        false
        8087:aaa        false
generic 1d6b:2  false
990chroma       1d6b:2  false
990     1d6b:2  false
913     1d6b:2  false
908     1d6b:2  false
715     1d6b:2  false
711     1d6b:2  false
709     1d6b:2  false
        1d6b:2  false
Warning: Unknown DPI value 100
Warning: Unknown DPI value 200
Warning: Unknown DPI value 300
Warning: Unknown DPI value 8000
Warning: Unknown DPI value 16000
[tol@tol-manjaro mouse_m908]$ ^C

```

</p>
</details>  

Result:

dokutan, 2021-03-04:
What you did looks good so far, however the binary you actually called is definitely not coming from these commits, i removed the debug output during mouse detection as that was solved. I don't know the details of how you upgraded but the general process that should work is, i suspect you are missing the ``make clean``:
```
cd path/to/mouse_m908
git pull
make clean
make
sudo make install  # if you leave this out, make sure to run the local binary with ./mouse_m908
```
If you try to build as described above and run the tests again the output should be a hexdump of the sent data.

anvena, 2021-03-04:
Great it works as expected:

[tol@tol-manjaro mouse_m908]$ mouse_m908 -c examples/example_m913.ini
0       : 08 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 49 
1       : 08 07 00 01 20 08 02 80 04 00 40 04 00 8b 00 00 c8 
2       : 08 07 00 01 40 08 02 81 20 00 41 20 00 51 00 00 a8 
3       : 08 07 00 01 80 08 02 81 22 00 41 22 00 4d 00 00 68 
4       : 08 07 00 01 a0 08 02 81 23 00 41 23 00 4b 00 00 48 
5       : 08 07 00 02 00 08 02 81 24 00 41 24 00 49 00 00 e7 
6       : 08 07 00 02 20 08 02 81 25 00 41 25 00 47 00 00 c7 
7       : 08 07 00 02 80 08 02 81 26 00 41 26 00 45 00 00 67 
8       : 08 07 00 02 a0 08 02 81 27 00 41 27 00 43 00 00 47 
9       : 08 07 00 02 c0 08 02 81 57 00 41 57 00 e3 00 00 27 
a       : 08 07 00 02 e0 08 02 81 56 00 41 56 00 e5 00 00 07 
b       : 08 07 00 00 60 08 01 01 00 53 01 10 00 44 00 00 34 
c       : 08 07 00 00 68 08 08 08 00 00 05 00 00 50 00 00 2c 
d       : 08 07 00 00 70 08 05 00 00 50 05 00 00 50 00 00 24 
e       : 08 07 00 00 78 08 05 00 00 50 05 00 00 50 00 00 1c 
f       : 08 07 00 00 80 08 05 00 00 50 05 00 00 50 00 00 14 
10      : 08 07 00 00 88 08 05 00 00 50 05 00 00 50 00 00 0c 
11      : 08 07 00 00 90 08 05 00 00 50 05 00 00 50 00 00 04 
12      : 08 07 00 00 98 08 05 00 00 50 05 00 00 50 00 00 fc 
13      : 08 07 00 00 0c 08 00 00 00 55 02 02 00 51 00 00 88 
14      : 08 07 00 00 14 08 03 03 00 4f 5f 5f 97 4d 00 00 80 
15      : 08 07 00 00 1c 04 bd bd 00 db 00 00 00 00 00 00 d1 
16      : 08 07 00 00 02 02 05 50 00 00 00 00 00 00 00 00 ed 
17      : 08 07 00 00 2c 08 ff 00 00 56 00 00 ff 56 00 00 68 
18      : 08 07 00 00 34 08 00 ff 00 56 ff ff 00 57 00 00 60 
19      : 08 07 00 00 3c 04 ff 55 7d 84 00 00 00 00 00 00 b1 
1a      : 08 07 00 00 54 08 ff 00 ff 57 03 52 80 d5 00 00 eb 
1b      : 08 07 00 00 5c 02 03 52 00 00 00 00 00 00 00 00 93 
1c      : 08 07 00 00 00 02 02 53 00 00 00 00 00 00 00 00 ef 
[tol@tol-manjaro mouse_m908]$ 

Only one issue button 3 function was not changed.

dokutan, 2021-03-04:
Glad that it worked now, buttons 13-16 not working is expected because i didn't implement them yet, as for button 3 could you try mapping it to something else, e.g. forward to see if the problem is a wrong keycode for led_toggle or a wrong button position?

dokutan, 2021-03-04:
The button positions were indeed mixed up, that should be fixed in 3829421f2be3ab6cbc77ab6d20cbeebc12aa8288. If it is now possible to remap all buttons i can focus on the button functions.

anvena, 2021-03-04:
I set button_3=forward and it works.

dokutan, 2021-03-04:
Perfect, does remapping the other buttons work?

anvena, 2021-03-05:
Remapping is working for all 1-12 buttons. What polling_rate is? I assigned to button but it did not override previous value.

dokutan, 2021-03-05:
> What polling_rate is? I assigned to button but it did not override previous value.

I will look into that and consider the other functions. What dou you plan do capture next, led settings or the remaining DPI values?

anvena, 2021-03-05:
Yes 13-16 also works fine. Next I'll capture dpi levels.

anvena, 2021-03-07:
Hi, I've captured all possible dpi value from 1000 to 16000 (from 100 to 100 it was already captured before). Also package 3 contains data fro mouse sensitivity and scroll speed.

Have a good day.

dokutan, 2021-03-07:
Thanks, i have now implemented all DPI values, however changing the sensitivity and scrollspeed doesn't appear as a change in the sent data. This probably means that these settings are implemented by the software running on windows and are not stored on the mouse, to confirm this you could try changing these settings and then see if there is a change when using the mouse on Linux.

anvena, 2021-03-07:
Yes you are right these settings somehow synced with windows settings.

dokutan, 2021-03-07:
If i didn't miss anything that leaves the LED related settings and reading the settings from the mouse unimplemented and requiring captured data.

anvena, 2021-03-11:
Would you mind describing what is a preferable steps to capture data ?

dokutan, 2021-03-11:
- Changing to each available lightmode, while not changing the color/speed/brightness if possible
- Changing brightness, lightmode and LED speed to every possible value
- Change to 2 different colors
- Capture while opening the software after closing it completeley from the taskmanager
- It would be helpful if you could add a screenshot of the lighting section in the software

I hope that answers your question

matthewbaggett, 2022-01-03:
Would it help accellerate working out how to write macros (and things like simply having the "fire" button trigger F13, etc) if I sent one of these mice to you? I would have sent an email, but I can't find an email address for @dokutan

dokutan, 2022-01-03:
> if I sent one of these mice to you? I would have sent an email, but I can't find an email address for @dokutan

@matthewbaggett Thank you for the offer, but there is no need as i already gained access to a M913. Currently my limitations are a lack of free time and an annoying protocol (e.g. checksums are used when mapping a button to keyboard keys).

The best thing is therefore to help with reverse engineering/documenting the protocol. The issue for the M686 (which shares the protocol) has more information: https://github.com/dokutan/mouse_m908/issues/29#issuecomment-1003439764

dokutan, 2022-01-03:
@matthewbaggett Mapping buttons to keyboard keys (including F13-F24) is implemented with aa26e9780697074dc0569e069a810fa04fb4c0b4. Key combinations and macros are still missing.

CodexHere, 2024-07-10:
I just got one of these mice and was curious if there is anything that'd help get RGB settings functional? I noticed in the example config it says it's not supported, and sure enough I didn't notice it take effect.

Dying to change the color on the fly (it helps me figure out my profile).

dokutan, 2024-07-10:
@CodexHere iirc, the message to change the RGB settings includes a checksum that is required to be correct for the settings to be applied. I started reverse engineering the rest of the protocol, see https://github.com/dokutan/mouse_m908/blob/master/include/m913/writers.cpp. You could try to figure out how this checksum works and implement it.

Philosoph228, 2025-01-05:
@dokutan it's freaking me out. I had the wired version of this mouse, and I became very dependent on binding DPI to VOL-/VOL+. Controlling media volume on my PC with just my fingers, like a TV remote, was absolutely divine. Since the white version of the wired M908 wasn't available, I ordered the wireless M918 instead.

Discovering its more limited functionality — having fewer profiles and, specifically, the inability to rebind the arrow keys and — has left me very upset and disappointed.

I believe the DPI button functionality is somehow hardcoded but could be changed by sending specific commands. I tried replacing the button IDs with 0x11/0x12 in the Cfg.ini file, but it didn't work. Since you have this mouse, do you know if it's possible to change the DPI button function? By sending specific commands or even perhaps by downloading the EEPROM via USB and modifying the firmware?

dokutan, 2025-01-06:
> @dokutan it's freaking me out. I had the wired version of this mouse, and I became very dependent on binding DPI to VOL-/VOL+. Controlling media volume on my PC with just my fingers, like a TV [...]

Unfortunately i don't know for sure because i don't have the M913 anymore, but iirc it is possible to change the behaviour of the DPI buttons with the official software. Therefore i would recommend setting up a Windows VM with USB passthrough/a Windows machine and the official software, since the settings are stored on the mouse this should only be required once.