dokutan/mouse_m908 #16: Will this work on M688-1 Wireless Model?

sandikata, 2021-03-22:
Hello, is that going to work for Reddragon Shark2 M688-1 Wireless Mouse ?

Bus 003 Device 015: ID 062a:415e MosArt Semiconductor Corp. 2.4G INPUT DEVICE

dokutan, 2021-03-22:
Considering that mouse has a different vendor id from the other mice, it most likely uses yet another different protocol, meaning it is not compatible without doing some reverse engineering. 

To implement support you would have to install the official software (virtual machine with USB passthrough or bare metal) and then use wireshark to capture the USB communication with the mouse. Here is my comment from a similar issue with more details: https://github.com/dokutan/mouse_m908/issues/15#issuecomment-779279326. It would also be useful to know if the mouse supports wired operation and the ``lsusb`` output if it does.

sandikata, 2021-03-23:
Well, it is only wireless and even more there's no software for such model, i tried with software for closest model it says unsupported hardware. I think this mouse it's just hardcoded not programable.

It has 3 buttons for Alt Shift and CTRL, another 2 for DPI +/-, one for Web Home Page, One for fast triple click, and eco mode switch. So it works out of the box. I cannot say if it's programable in any way as there's no any redragon driver or software for M688 series.

closed 2021-03-23 by sandikata