dokutan/mouse_m908 #18: M990 RGB Legend Chroma Could not read fire button config

grafov, 2021-03-25:
I did not try to reconfigure mouse buttons yet, the new mouse has default configuration. When I tried to read configuration with `mouse_m908 -R -` it displayed:

```
button_fire=unknown, please report as bug:  92  3  1  0
```

for all the profiles. Other buttons reported without errors. The fire button by default works like left button (and `xev` reported it as "button 1").

The full output of the commands: 
* [mouse-m990-rgb-read.txt](https://github.com/dokutan/mouse_m908/files/6206555/mouse-m990-rgb-read.txt)
* [mouse-m990-rgb-dump.txt](https://github.com/dokutan/mouse_m908/files/6206556/mouse-m990-rgb-dump.txt)

dokutan, 2021-03-25:
Thanks for the report, this is indeed an alternative code for the fire button, i have added support in 40eef1cf1ba70ed9409029596ba4832bfb5b4404.

Please note that if you are interested in configuring all buttons of the mouse and reading the configuration of the remaining buttons, you would need to capture the USB communication from the official software with wireshark to allow me to implement these features.

grafov, 2021-03-25:
Thank you for the reply! It would be great to get the fully working configuration support on Linux. I have already installed official software to the Windows machine, I'll install Wireshark too. Could you provide the instructions that exactly I need to capture?

dokutan, 2021-03-25:
A few links about the process copied from a previous issue:
https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc (from my other usb reverse engineering project)

The idea is to capture:
- the data sent when changing each (unsupported) button on its own to a known value, e.g. a specific keyboard key. It would additionally be helpful if you could attach a screenshot or list of what all buttons do for each capture.
- the data sent when opening the official sofware after closing it completely from the task manager

REmerald, 2022-06-21:
Hi, I setted up the reverse engineering environment in windows pe, installed the official software, wireshark, setted up the filter (`usb.bus_id == 1 && usb.device_address == 2 && usb.src == "host"`), etc.
When I change the mouse settings in the software, I get the entries/frames in wireshark with the hex values, how to "understand" them?
When I change the settings that apply immideately, I get 6 entries in wireshark. I noticed that the 2nd one contains the hex of the color I set. This is the only thing I understood.
But some settings don't apply immideately and need clicking the "apply" button to actually apply, so when I click the button, it applies all the settings and I get around 175 entries in wireshark. This makes me feel completely lost.
So what should I do with the output?
I can send it here. Here is the output of launching and quiting from the official software (only launching prodused any output). It contains output from the other devices, so don't forget to apply the filter above, when you'll read it.
[m990-rgb--bus-1--addr-2_software-launch-quit.pcapng.gz](https://github.com/dokutan/mouse_m908/files/8950428/m990-rgb--bus-1--addr-2_software-launch-quit.pcapng.gz)

REmerald, 2022-06-21:
Btw, here is the output of changing the mode/profile by the software (instead of by the special button on the mouse):
[image]
[image]
[m990-rgb--bus-1--addr-2_mode1-to-mode2.pcapng.gz](https://github.com/dokutan/mouse_m908/files/8950995/m990-rgb--bus-1--addr-2_mode1-to-mode2.pcapng.gz)
[m990-rgb--bus-1--addr-2_mode2-to-mode1.pcapng.gz](https://github.com/dokutan/mouse_m908/files/8950997/m990-rgb--bus-1--addr-2_mode2-to-mode1.pcapng.gz)
Applies immediately right after selecting the mode from the combobox.
Maybe this would be a cool feaure to implement.

dokutan, 2022-06-21:
> When I change the mouse settings in the software, I get the entries/frames in wireshark with the hex values, how to "understand" them?

Understanding them means which bytes need to be set (and to what values, for speeds etc.) for each setting. e.g. which bytes represent the LED color.

> But some settings don't apply immideately and need clicking the "apply" button to actually apply, so when I click the button, it applies all the settings and I get around 175 entries in wireshark. This makes me feel completely lost.
So what should I do with the output?

I find it helpful to only change one setting at a time, then export the captured packets as json. This allows you to narrow down the changed bytes using tools like ``grep`` and ``diff``. You can see that a lot of packets are sent each time the settings are changed in https://github.com/dokutan/mouse_m908/blob/master/include/m990chroma/data.cpp, but only a few bytes are set to different values in https://github.com/dokutan/mouse_m908/blob/master/include/m990chroma/writers.cpp. If you don't want to do that, just send me your captured data.

> Maybe this would be a cool feaure to implement.

This should alread work using the ``-p`` option, please let me know if that is not the case.

I will have a look at remapping buttons 13-16. It would be extremely helpful if you could confirm which other features of mouse_m908 work or don't work for the M990 Chroma.

dokutan, 2022-06-21:
@REmerald Which and how many front buttons can be customized?

REmerald, 2022-06-22:
Reading the settings from the mouse seems alright, except the macros section (and >12 buttons ofc):
```
# Macros

;## macro8
;# unknown, please report as bug: ff ff 0
```
I didn't make any macros in the official software as I remember. Here's an ini file and a dump in case you might need it:
[m990rgb_remerald.ini.txt](https://github.com/dokutan/mouse_m908/files/8957409/m990rgb_remerald.ini.txt)
[m990rgb_remerald.dump.txt](https://github.com/dokutan/mouse_m908/files/8957410/m990rgb_remerald.dump.txt)

REmerald, 2022-06-22:
I modified the ini file
[m990rgb_myconf.ini.txt](https://github.com/dokutan/mouse_m908/files/8957570/m990rgb_myconf.ini.txt)
and it completely messed up my mouse settings: the only thing that worked is setting the rgb color and (maybe) brightness in mode 1. The dpi became the same on all modes: mid-high on X and very low on Y. 1-12 buttons didn't change, 13 became ledmodeswitch on all modes except 1, 14 became change mode on all profiles. 15 and 16 didn't change. The lowest button on the top of the mouse became dpicycle (or dpi+?) on all modes except 1.
Maybe that's because I added `button_13` in the ini file.
Then I read the settings again:
[m990rgb_messedup.ini.txt](https://github.com/dokutan/mouse_m908/files/8957774/m990rgb_messedup.ini.txt)
[m990rgb_messedup.dump.txt](https://github.com/dokutan/mouse_m908/files/8957775/m990rgb_messedup.dump.txt)

REmerald, 2022-06-22:
Here's the version I used:
```
 remorin@archsys@pts/2 ~ $ sudo mouse_m908 -v                                                                          4:03PM | Wed, Jun 22
Version: 3.3
 remorin@archsys@pts/2 ~ $ paru -Sii mouse_m908                                                                        4:05PM | Wed, Jun 22
Repository      : aur
Name            : mouse_m908
Version         : 3.3-1
Description     : Control Redragon gaming mice from Linux, BSD and Haiku
URL             : https://github.com/dokutan/mouse_m908
AUR URL         : https://aur.archlinux.org/packages/mouse_m908
Groups          : None
Licenses        : GPL3
Provides        : mouse_m908
Depends On      : libusb  gcc-libs
Make Deps       : make
Check Deps      : None
Optional Deps   : None
Conflicts With  : None
Maintainer      : dokutan
Votes           : 3
Popularity      : 0.034016
First Submitted : Tue, 03 Aug 2021 16:49:03 +0000
Last Modified   : Sat, 22 Jan 2022 01:36:31 +0000
Out Of Date     : No
ID              : 1023172
Package Base ID : 169560
Keywords        : None
Snapshot URL    : https://aur.archlinux.org/cgit/aur.git/snapshot/mouse_m908.tar.gz

```

dokutan, 2022-06-22:
@REmerald For testing purposes, building the current git version is required (the reported version is still 3.3):
```
git clone https://github.com/dokutan/mouse_m908 && cd mouse_m908
# or
git pull

make clean
make
sudo make install
```

I updated setting the button mapping in commit 0f510f3e047b2af35a53e834bb899d332c71f158.