dokutan/mouse_m908 #19: Redragon M719 Invader

ma1uta, 2021-04-03:
I bought a mouse "Redragon M719 Invader" and read config with command `mouse_m908 -R mouse.ini`

Default config in the attached file.

Hope this will be helpful. If I can help with something, just ask me :)

[mouse.txt](https://github.com/dokutan/mouse_m908/files/6253115/mouse.txt)

ma1uta, 2021-04-03:
I applied config `mouse_m908 -c mouse.txt` 
[mouse.txt](https://github.com/dokutan/mouse_m908/files/6253231/mouse.txt)

Then read mouse settings `mouse_m908 -R mouse.result.txt`
[mouse.result.txt](https://github.com/dokutan/mouse_m908/files/6253232/mouse.result.txt)

dokutan, 2021-04-03:
Thanks, at least this confirms that the USB ids are correct and the protocol is compatible. But apart from an alternative code for the ``wave`` lightmode there is not much i can read from this alone.

It would further be helpful if you could attach the outout of ``mouse_m908 -D -`` and a list of the buttons the mouse has. If you are interested in implementing full support it would be neccessary to capture the usb communication from the official software with wireshark, please ask if you need more details.

ma1uta, 2021-04-03:
Output of the `mouse_m908 -D -` in the attached file.
[dump.txt](https://github.com/dokutan/mouse_m908/files/6253243/dump.txt)

Buttons:
- left
- right
- scroll wheel (clicked as middle button)
- rapid fire button (default 3 rapid clicks)
- side key, default forward
- side key, default backward
- dpi button to switch DPI 500-1000-2000-3000-5000
- light mode toggle key

At bottom there is a user mode toggle key

User manual:
[M719_User_Manual_05f5490f-8fc1-4aef-8225-2906cc0dc710.pdf](https://github.com/dokutan/mouse_m908/files/6253251/M719_User_Manual_05f5490f-8fc1-4aef-8225-2906cc0dc710.pdf)

I download software https://www.redragonzone.com/pages/download and can try run it with wine for example. Could you please explain how to capture the usb communication.

dokutan, 2021-04-03:
Unfortunately running it in wine will not work as direct usb access is needed. You can use a virtual machine with usb passthrough (i use qemu/kvm, virtualbox should work as well) or a native windows installation.

Here are a few guides on the capturing process, you can use wireshark both on the host in the case of a vm or on windows.
https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc (from my other usb reverse engineering project)

I hope this helps, feel free to ask if there is anything else.

ma1uta, 2021-04-03:
Ok, I will try run windows via virtualbox/qemu with usb passthrough. What does I should try with mouse? Record each clicks in the separated file?

dokutan, 2021-04-03:
- Capture while applying the settings, please add screenshot/note what you applied.
- Capture while opening the software after completely closing it from the taskmanager.

dokutan, 2021-04-03:
I have now added a backend for the M719, if you update with
```
git pull
make clean
make
sudo make install
```
and run ``mouse_m908 -R -`` the detected model should be ``719`` instead of ``generic``.

If that works and you are finished with the capturing process i can begin implementing the actual features.

ma1uta, 2021-04-03:
Yes, It works, output of the `mouse_m908 -R -` in the attached file
[mouse_719.txt](https://github.com/dokutan/mouse_m908/files/6253746/mouse_719.txt)

I'll finish capturing process tomorrow, just downloaded windows 7.

ma1uta, 2021-04-04:
Open settings:
[opensettings.zip](https://github.com/dokutan/mouse_m908/files/6254347/opensettings.zip)

How settings looks:
[image]

ma1uta, 2021-04-04:
When click on any button popups the menu:

[image]
[image]
[image]
[image]
[image]
[image]

ma1uta, 2021-04-04:
Setup 6th button to dpi cycle.

[setup6toDpiCycle.zip](https://github.com/dokutan/mouse_m908/files/6254386/setup6toDpiCycle.zip)

https://user-images.githubusercontent.com/1460469/113506472-a42a3680-9534-11eb-8e1f-edc09bbe73e1.mp4

ma1uta, 2021-04-04:
Setup 6th button to `tilde`

[image]

[setup7toTilde.zip](https://github.com/dokutan/mouse_m908/files/6254412/setup7toTilde.zip)

https://user-images.githubusercontent.com/1460469/113506678-ca040b00-9535-11eb-9799-c9e73d50b5fe.mp4

ma1uta, 2021-04-04:
How look the Fire button settings:
[image]

ma1uta, 2021-04-04:
I hope this will be helpful. If you need I can record other scenarios. :)

ma1uta, 2021-04-04:
How look macro manager.
[image]

dokutan, 2021-04-04:
Thanks, this was very helpful, especially the detailed screenshots, also nice awesomewm setup ;) . Both the software and the protocol are indeed identical with the M908 for the most part, only the different amount of buttons is obvious.

I have used your data to improve the process of applying the settings in commit 6450bee328d14fe571aac8898dc4a87d6e562893, you can now try to change the various settings including remapping the buttons and macros. Please tell me what works and what doesn't.

As for reading of the settings, unfortunately your capture didn't include any useful data. I suspect that the software was still running in the background somehow or it doesn't implement that feature like this. You could try capturing while logging in to windows and use the wireshark display filter to confirm that there is communication with the mouse.

ma1uta, 2021-04-04:
Unfortunately, buttons remapping still not works:

I run:
```
git pull
make clean
make
sudo make install
```
then reboot the OS,

I tried load `mouse_m908 -c mouse.new.txt` with config in the attached file
[mouse.new.txt](https://github.com/dokutan/mouse_m908/files/6255180/mouse.new.txt)

Result of the `mouse_m908 -R '` in the attached file:
[output.txt](https://github.com/dokutan/mouse_m908/files/6255183/output.txt)

In the settings program I see:
[image]

DPI values:
[image]
100 - 0x0200
200 - 0x0400
300 - 0x0600
400 - 0x0800
500 - 0x0b00

[image]
600 - 0x0d00
700 - 0x0f00
800 - 0x1200
900 - 0x1400
1000 - 0x1600

[image]
1100 - 0x1900
1200 - 0x1b00
1300 - 0x1d00
1400 - 0x2000
1500 - 0x2200

(then without screenshots to increase speed)
1600 - 0x2400
1700 - 0x2700
1800 - 0x2900
1900 - 0x2b00
2000 - 0x2e00
2100 - 0x3000
2200 - 0x3200
2300 - 0x3400
2400 - 0x3700
2500 - 0x3900
2600 - 0x3b00
2700 - 0x3e00
2800 - 0x4000
2900 - 0x4200
3000 - 0x4500
3100 - 0x4700
3200 - 0x4900
3300 - 0x4c00
3400 - 0x4e00
3500 - 0x5000
3600 - 0x5300
3700 - 0x5500
3800 - 0x5700
3900 - 0x5a00
4000 - 0x5c00
4100 - 0x5e00
4200 - 0x6100
4300 - 0x6300
4400 - 0x6500
4500 - 0x6800
4600 - 0x6a00
4700 - 0x6c00
4800 - 0x6f00
4900 - 0x7100
5000 - 0x7300

After 5000 step increase from 100 to 200:
5200 - 0x3b01
5400 - 0x3e01
5600 - 0x4001
5800 - 0x4201
6000 - 0x4501
6200 - 0x4701
6400 - 0x4901
6600 - 0x4c01
6800 - 0x4e01
7000 - 0x5001
7200 - 0x5301
7400 - 0x5501
7600 - 0x5701
7800 - 0x5a01
8000 - 0x5c01
8200 - 0x5e01
8400 - 0x6101
8600 - 0x6301
8800 - 0x6501
9000 - 0x6801
9200 - 0x6a01
9400 - 0x6c01
9600 - 0x6f01
9800 - 0x7101
10000 - 0x7301

Also, could you please check this captured image:
[settings.zip](https://github.com/dokutan/mouse_m908/files/6255243/settings.zip)
I run:
- start capture usb
- enable usb passthrought in the virt-manager
- see a lag and the program see current settings
- stop capture

dokutan, 2021-04-04:
About the button remapping, reading the current mapping from the mouse is not yet implemented but changing works independently. When you use the following section in your config, do the left and right mouse buttons get swapped?
```
[profile1]
button_left=right
button_right=left
```
And just fyi, you don't have to reboot after updating the program.

dokutan, 2021-04-04:
Actual DPI values are implemented in commit 62e157280d094a737a1d795d3ebc4598822e934f. The last capture looks promising as well.

ma1uta, 2021-04-04:
> When you use the following section in your config, do the left and right mouse buttons get swapped?

No, not swapped but scroll_down and scroll_up buttons will scroll up both.

Also `mouse_m980 -R -` prints the following config:
```
button_forward=dpi-cycle
button_backward=forward
button_fire=backward
```
But the correct config should be:
```
button_fire=dpi-cycle
button_forward=forward
button_backward=backward
```

`mouse_m980 -R -` prints correct DPI values, thanks.

dokutan, 2021-04-04:
It looks like the order of the buttons was wrong, this should hopefully be fixed by 4c099a1b5298aa874985f28085b838d813f5cebc.

> mouse_m980 -R - prints correct DPI values, thanks.

Perfect, thanks for letting me know. Do the led related settings work as intended?

ma1uta, 2021-04-05:
Led switch and color settings works.

Config:
```
button_left=left
button_right=right
button_middle=middle
button_fire=dpi-cycle
button_forward=forward
button_dpi=backward
button_lightmode=led_mode_switch
button_backward=Tilde
```

Will be looks 
[image]

You can see that `button_dpi` and `button_backward` are swapped.
And scroll_down works as scroll_up.

And `mouse_m908 -R -` doesn't output scroll_up and scroll_down buttons.

dokutan, 2021-04-05:
The swapped buttons are corrected by 28578e948861ade8d5d755595d9619990a64eb04.

Unfortunately it doesn't seem possible to read the mappings of ``scroll_up`` and ``scroll_down``, they don't appear in the captured data and the fact that the official software displays the wrong values supports this further. What i don't quite understand yet is if remapping works or not. How does the mouse behave after
```
[profile1]
scroll_up=scroll_up
scroll_down=scroll_down
```
and
```
[profile1]
scroll_up=scroll_down
scroll_down=scroll_up
```
?

If this doesn't work, but the official software is capable of remapping the scrollwheel, could you capture the data when changing the scrollwheel to two different mappings?

ma1uta, 2021-04-05:
In both cases
```
[profile1]
scroll_up=scroll_up
scroll_down=scroll_down
```
and
```
[profile1]
scroll_up=scroll_down
scroll_down=scroll_up
```

After `mouse_m908 -c mouse.ini` scroll_up == scroll_down and works as scroll_up both.

In the attached file:
[scrolls.zip](https://github.com/dokutan/mouse_m908/files/6260132/scrolls.zip)

First in the `scrollReverse` file: setup scroll_up=scroll_down and scroll_down=scroll_up
After that `scrollDefault` file: setup scroll_up=scroll_up and scroll_down=scroll_down

dokutan, 2021-04-05:
There was a bug that i am suspecting to be the cause of the issue, that got fixed by bdf5c59bad2f59d536dc6422df23272c20e849e. I hope it works now but am continuing to look if there is anything else that could be the reason for this.

ma1uta, 2021-04-06:
Yeah, scrolls works as expected! Thanks.

Config
```
[profile1]
scroll_up=scroll_down
scroll_down=scroll_up
```
reversed scroll buttons.

```
[profile1]
scroll_up=scroll_up
scroll_down=scroll_down
```
Works correct too.

ma1uta, 2021-04-06:
Macro manager. I will create a few macros and write a message with a screenshot of the macros and captured data when save macros.

ma1uta, 2021-04-07:
First macro:
[image]
[image]

Captured data:
[test1Macro.zip](https://github.com/dokutan/mouse_m908/files/6273360/test1Macro.zip)

`mouse_m908 -R -` shows:
```
# Macros

;## macro1
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left
;# down mouse_right
;# up   mouse_right
```

I tried setup the second macro with the official program but it seems have bugs. I could not force work two or more macros even with official program normally.
So, I think we can stop on this. Buttons, remapping, led controls work perfect.
Many thanks for this, you done a wonderful work.
I can close this issue if you want.

dokutan, 2021-04-07:
The last thing that might be worth trying out is if that single macro works properly, after all having one macro is better than having no macros. I disabled the reading and writing of all macros except macro1 in 859dfb703bc28cba96a4ea2ab157a8a4f07df340.

You could change ``examples/example_m719.ini`` to map any button to ``macro1`` then run
```
mouse_m908 -c examples/example_m719.ini
mouse_m908 -m examples/example_m719.ini
mouse_m908 -R -
```

to check if it is at least possible to change the single macro.

If that works i would mark the M719 as completely supported in the README and then close this issue.

ma1uta, 2021-04-08:
Example from `example_m719.ini` works (read macros and write macros) correct without issues.

I tried the next macros:
```
;## macro1
;# down mouse_right
;# up   mouse_right 
```
And it doesn't work.

dokutan, 2021-04-08:
Can you tell me what you did and where something fails?

ma1uta, 2021-04-11:
I have config:
```
# Model: 719
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# Note: reading the scrollspeed is not supported by the mouse.

# Currently active profile: 1

[profile1]

# LED settings
color=ffffff
brightness=3
speed=0
lightmode=off
report_rate=500

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=800
dpi2_enable=1
dpi2=1500
dpi3_enable=0
dpi3=10000
dpi4_enable=0
dpi4=9600
dpi5_enable=0
dpi5=9800

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=dpi-cycle
button_forward=forward
button_backward=backward
button_lightmode=Tilde
button_dpi=macro1
scroll_up=scroll_up
scroll_down=scroll_down

[profile2]

# LED settings
color=ffffff
brightness=3
speed=1
lightmode=unknown, please report as bug: 02 01
report_rate=125

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=scroll_up
button_right=scroll_up
button_middle=scroll_up
button_fire=scroll_down
button_forward=left
button_dpi=right
button_lightmode=middle
button_backward=fire:mouse_left:3:0

[profile3]

# LED settings
color=ffffff
brightness=3
speed=1
lightmode=unknown, please report as bug: 02 01
report_rate=125

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=forward
button_right=dpi-cycle
button_middle=dpi-cycle
button_fire=forward
button_forward=forward
button_dpi=backward
button_lightmode=scroll_up
button_backward=scroll_down

[profile4]

# LED settings
color=ffffff
brightness=3
speed=1
lightmode=unknown, please report as bug: 02 01
report_rate=125

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=middle
button_forward=fire:mouse_left:3:0
button_dpi=forward
button_lightmode=forward
button_backward=dpi-cycle

[profile5]

# LED settings
color=ffffff
brightness=3
speed=1
lightmode=unknown, please report as bug: 02 01
report_rate=125

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=forward
button_right=backward
button_middle=scroll_up
button_fire=scroll_down
button_forward=scroll_down
button_dpi=left
button_lightmode=right
button_backward=middle

# Macros

;## macro1
;# down mouse_right
;# up   mouse_right 
```

Then run:
```
mouse_m908 -c mouse.new.ini
mouse_m908 -m mouse.new.txt
```
And nothing happens.

```
mouse_m908 -R -
```

shows config:
```
# Model: 719
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# Note: reading the scrollspeed is not supported by the mouse.

# Currently active profile: 1

[profile1]

# LED settings
color=ffffff
brightness=3
speed=0
lightmode=off
report_rate=500

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=800
dpi2_enable=1
dpi2=1500
dpi3_enable=0
dpi3=10000
dpi4_enable=0
dpi4=9600
dpi5_enable=0
dpi5=9800

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=dpi-cycle
button_forward=forward
button_backward=backward
button_lightmode=Tilde
button_dpi=macro1
# reading scroll_up and scroll_down is not supported

[profile2]

# LED settings
color=ffffff
brightness=3
speed=2
lightmode=static
report_rate=125

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=scroll_up
button_right=scroll_up
button_middle=scroll_up
button_fire=scroll_up
button_forward=scroll_down
button_backward=left
button_lightmode=fire:mouse_left:3:0
button_dpi=middle
# reading scroll_up and scroll_down is not supported

[profile3]

# LED settings
color=ffffff
brightness=3
speed=2
lightmode=static
report_rate=125

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=right
button_right=none
button_middle=forward
button_fire=dpi-cycle
button_forward=dpi-cycle
button_backward=forward
button_lightmode=forward
button_dpi=scroll_down
# reading scroll_up and scroll_down is not supported

[profile4]

# LED settings
color=ffffff
brightness=3
speed=2
lightmode=static
report_rate=125

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=scroll_up
button_right=backward
button_middle=none
button_fire=left
button_forward=right
button_backward=middle
button_lightmode=middle
button_dpi=fire:mouse_left:3:0
# reading scroll_up and scroll_down is not supported

[profile5]

# LED settings
color=ffffff
brightness=3
speed=2
lightmode=static
report_rate=125

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=1
dpi1=200
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=5000

# Button mapping
button_left=dpi-cycle
button_right=forward
button_middle=forward
button_fire=none
button_forward=forward
button_backward=backward
button_lightmode=scroll_up
button_dpi=scroll_down
# reading scroll_up and scroll_down is not supported

# Macros

```

ma1uta, 2021-04-11:
It seems mouse button doesn't works.

When I try config from the example (print "macro1") it works.

ma1uta, 2021-04-11:
Sorry, It's my fault, I don't uses '\t' in the macros. Let me check this.

ma1uta, 2021-04-11:
Yes, everything works if using the `\t` character. `mouse_m908 -R -` shows only the first macros but reading and writing macros works correct.

I think we close this issue because all works perfect. Thanks for adding support for this mouse!

dokutan, 2021-04-11:
Perfect, i just updated the README and am now closing this issue. If you discover anything new, bugs or features in the official software, feel free to open a new issue.

FYI for the M908, there are two versions of the official software, one that looks like your screenshots and a newer one that added a few minor features. So it is be possible that a new version for the M719 gets released that fixes the bugs (inability to read the mapping of the scrollwheel, broken macros), if that happens it should be possible to fix them here as well.

And thank you for providing this detailed information about the mouse, it wouldn't have been possible to implement support without you.

closed 2021-04-11 by dokutan