dokutan/mouse_m908 #20: help how do i get it back to rainbow

Gustiviy, 2021-04-12:
m908

dokutan, 2021-04-12:
I'm not sure what exactly your question and problem is. At least the commands you ran, what happened and what you want to happen would be appropriate.

If you want to change the lightmode to rainbow, edit your config.ini file (that you can obtain from the examples directory or by running ``mouse_m908 -R config.ini``) to say
```
[profile1]
# or [profile2]-[profile5]
lightmode=rainbow
```
then run ``mouse_m908 -c config.ini``. To change the active profile, run e.g. ``mouse_m908 -p 1``. I would further recommend having a look at https://github.com/dokutan/mouse_m908/blob/master/examples/example_m908.ini.

dokutan, 2021-04-19:
As the question seems to be answered, i am closing this issue.

closed 2021-04-19 by dokutan