dokutan/mouse_m908 #21: M702 Phoenix support

navid-zamani, 2021-04-19:
Hello, I have an M702 mouse, and would like to request support.

What information do you need?
(I’m good at Linux and programming, and will help wherever I can, so no need to simplify. :)

navid-zamani, 2021-04-24:
Hey, thanks!
The device IDs are 04d9:fc15.

Somehow I can't add them myself. Added the correct pid to include/generic/data.cpp:_c_all_pids, but it still doesn't detect the mouse, no matter what parameters I give it when I run the fully recompiled program.

Could you add it? I quadruple-checked the vid and pid.

dokutan, 2021-04-24:
I have added the PID in b9c8fbae3e9723e1858c333d26ebc1be5bfd37e9, it should only be required to add it like you presumably did. If it still doesn't work, please post the output of ``lsusb -vd 04d9:fc15``. Just to make sure you didn't miss something (i assume you already did these things):
- run ``make clean`` before recompiling
- run ``sudo make install`` or run the recompiled version from the local directory ``./mouse_m908 …``
- try running as root, ``sudo mouse_m908 …``

navid-zamani, 2021-04-24:
Okay, now it worked to ”read the data“, using `-R`. Apparently, while I already had tried everything you suggested, and even altered the code, I also had used `--kernel`, and it only works without. :)

It only read zeroes though.
So I won’t try changing anything, as resetting the mouse settings under Windows would be a major hassle.

I would now try all the protocols, but there is no switch for it. So I must refer back to you.

dokutan, 2021-04-24:
> I would now try all the protocols, but there is no switch for it. So I must refer back to you.

Yes, that would be the best, as your results show that the M702 is not compatible with the protocol implemented by the generic backend. Sadly it is not (yet) possible to easily switch between protocols as only one is properly understood, let me summarize what i know about the different protocols:
- there are at least 3 different protocols in use:
  - the M913 has a unique protocol (and a different vendor id)
  - the M990 has another unique protocol that is not completely understood
  - the M801 is not compatible with the generic backend as well, but no details about the protocol are known 
  - the most common one, used by the remaining supported mice and implemented by the generic backend

Therefore i would assume that the M702 and the M990 (and maybe the M801) share one protocol. You could try to change the product id in ``include/m990/data.cpp`` and then try to change the profile (reading the settings is not implemented for the M990) to check this or proceed directly to the next step.

In any case, given this result, it will be neccessary to capture the USB communication of the official software. Here are a few guides in case you are not familiar with the process:
https://bytepunk.wordpress.com/2017/03/25/reverse-engineering-a-usb-mouse/
https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol
https://github.com/dokutan/rgb_keyboard/tree/master/doc#capturing-usb-data

If i know whether the protocol is compatible between the M990 and M702 or have the captured data when changing the settings/profile i can add a backend for the M702 as a starting point for further reverse engineering of the protocol. I hope this helps, feel free to ask if there is anything else.