dokutan/mouse_m908 #25: Reverse engineering Redragon M607 Griffin

GrbavaCigla, 2021-07-30:
I am writing open source redragon driver (https://github.com/GrbavaCigla/opendragon) and I need help with reverse engineering. I am willing to contribute the .ini file when I figure it out.

Here are the pcap files I captured:
[m607.zip](https://github.com/dokutan/mouse_m908/files/6907750/m607.zip)

dokutan, 2021-07-30:
I don't quite understand what your goal is, it seems you are trying to write a kernel driver, which is something that should not be necessary if you want to implement the features provided by the official software (what this software does from the userspace). Only if the M607 doesn't even work as an input device, i would suspect an issue with the existing USB HID driver that should be fixed upstream.

To add proper support for the M607 more information would be required. From looking at your pcap files the protocol seems to be compatible with the generic implementation that should already support some features of the M607 (changing the profile, led settings might work), therefore i suggest that you try to change the settings with this software and then capture data for features that don't work. Additionally a list of all remappable buttons and the valid DPI range would be helpful.

That information would allow me to create a backend specific for the M607 (have a look at the directories in https://github.com/dokutan/mouse_m908/tree/master/include for the existing backends), which is what determines the actual support for the mouse, not the .ini file.

GrbavaCigla, 2021-07-30:
> I don't quite understand what your goal is, it seems you are trying to write a kernel driver, which is something that should not be necessary if you want to implement the features provided by the official software (what this software does from the userspace).

I could use userspace application but I want to learn how linux drivers work and linux in general :)

> which is what determines the actual support for the mouse, not the .ini file.

Yeah, I didn't read the source code, sorry. 😅

I captured more functions:
[m607.zip](https://github.com/dokutan/mouse_m908/files/6908314/m607.zip)

[image]
[image]

GrbavaCigla, 2021-07-30:
I just read the source code and I noticed hex values look like mine captures. Is there any meaning in those bytes? Looking at openrazer source code, they made a struct of 8 bytes and every byte has it's meaning.

dokutan, 2021-07-30:
> I just read the source code and I noticed hex values look like mine captures. Is there any meaning in those bytes? Looking at openrazer source code, they made a struct of 8 bytes and every byte has it's meaning.

If you mean the arrays in the various ``data.cpp`` files, then yes. The functions in ``writers.cpp`` copy the array and modify the copy to include the settings before sending them to the mouse. Of course using a struct (or class) to represent the transferred data might be a better approach, but that is how i ended up doing it. https://github.com/dokutan/mouse_m908/blob/master/documentation/default-annotated_m908.pdf has a description for the M908 if you are interested.

dokutan, 2021-07-30:
I have added an experimental backend for the M607 and included your captured data where it differs in 4ad551120586d1335d976d73bce51e4d0e560661. It would be great if you could build from the master branch and then post the output of ``mouse_m908 -R -`` and try to change the active profile with e.g. ``mouse_m908 -p 3``.

Additionally you could even try to change the settings with e.g. ``mouse_m908 -c examples/example_m607.ini`` and report which features work and which are broken. If you do this, i recommend that you have a second mouse available so you can use the official software to restore the settings, in case the buttons don't work afterwards.

GrbavaCigla, 2021-07-31:
I will check it soon. I found that it is not necessary to send all of the requests! I sent one of six requests and it worked! Their software seem buggy...

I simply sent this:
```c
unsigned char data[] = {0x02, 0xf1, 0x02, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
```
And mouse turned blue!

dokutan, 2021-07-31:
> Their software seem buggy...
True, and this isn't even the worst i've seen. It isn't possible to read a few settings on some mice, or only a single macro is supported, or …

The data you sent isn't even the "correct" one for changing the color. If you look at https://raw.githubusercontent.com/dokutan/mouse_m908/master/documentation/default-annotated_m908.pdf i would expect a 3 byte color value where only 0x00 0x00 0x00 are, (and the header is completely different). Just weird that it has this effect.

GrbavaCigla, 2021-07-31:
> The data you sent isn't even the "correct" one for changing the color.

Yes, I noticed a lot of reports cause mouse to go blue for 5 seconds (maybe upload mode or something?), but I think I found data fragment for changing to wave mode. I am going to investigate this a little more.

GrbavaCigla, 2021-08-04:
I tried your program now on my m607 and it seems to work.

dokutan, 2021-08-04:
Great, have you tried to remap the buttons succesfully? I would update the README if that works.

GrbavaCigla, 2021-08-05:
I tried example_m607.ini and remapping seems to work because DPI and color buttons are switched.

dokutan, 2021-08-05:
Thanks for the info, that should be fixed by aba3064eada9fc64eeb0ee29f9b11722011f565d. If everything else works as it should and you have no questions, feel free to close this issue.

GrbavaCigla, 2021-08-06:
OK, thank you for your help and info, my driver can finally change colors of the mouse :)

closed 2021-08-06 by GrbavaCigla