dokutan/mouse_m908 #29: Redragon M686 Support

MaxHarwell, 2021-10-16:
I am using a M686 Vampire Elite mouse and I am inquiring about possible support for this mouse and am willing to help contribute, I play Roblox daily and do general web browsing, so this mouse sees frequent use on my part.

dokutan, 2021-10-17:
Please run ``lsusb`` and attach the output, with that information i can add initial support for the M686. This would allow you to try some simple things like changing the profile to check if the protocol is compatible before working on the model specific features. 

For this later stage any information about the mouse would be helpful, e.g. a list of buttons, the DPI range, screenshots of the official software if possible.

MaxHarwell, 2021-10-17:
As im still somewhat new to linux, i will run into issues that may seem obvious when it comes to fixes, but i just ran into the error of libusb-1.0 was not found in the pkg-config search path and libusb.h does not exist when running the make command.

MaxHarwell, 2021-10-17:
```
Couldn't open mouse.
- Check hardware and permissions (maybe you need to be root?)
- Try with or without the --kernel-driver option
- Try with the --model option
- Try with the --bus and --device options
If nothing works please report this as a bug.
```
the software link worked, this is what i got when running `mouse_m908 -R -`

dokutan, 2021-10-17:
Try rebooting to load the udev rule or run as root: ``sudo mouse_m908 -R -``

MaxHarwell, 2021-10-17:
sudo worked and it identified it as the correct model.

dokutan, 2021-10-17:
Could you post the output and tell me what features where handled correctly?

dokutan, 2021-10-17:
That is not exactly what i had expected, try the same with ``sudo mouse_m908 -D -`` please.

MaxHarwell, 2021-10-17:
ah apologies, here you go.
```
09 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 48 
09 01 00 00 00 04 68 a8 3b 90 00 00 00 00 00 00 6c 
09 01 00 00 00 04 68 a8 3b 90 00 00 00 00 00 00 6c 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 04 00 00 00 02 09 01 00 00 00 00 00 00 00 00 3c 
```

MaxHarwell, 2021-10-17:
if theres anything you can do to get me started with looking at the traffic, that would be great, and would screenshots of what the default profile is help out in any way either?

dokutan, 2021-10-17:
> if theres anything you can do to get me started with looking at the traffic

You are looking for traffic on the wrong interface, load the usbmon kernel module (``sudo modprobe usbmon``) then choose usbmon1 in wireshark if the mouse is on bus 1. Additionally i would recommend removing either the wireless receiver or the usb cable to make sure the data gets send over the connection that you are actually capturing on.

> would screenshots of what the default profile is help out in any way either?

Absolutely yes, especially of the available buttons, LED effects and DPI.

MaxHarwell, 2021-10-17:
do you have anything besides this so we can communicate easier like discord? if you do my user is Max Harwell#2648

MaxHarwell, 2021-10-17:
i have wireshark open as root and the modprobe command doesnt do anything

MaxHarwell, 2021-10-17:
i set the display filter to `usb.bus_id==1&&usb.device_address==6`, which is the device for the mouse in question. now what? just poke around with the device software?

MaxHarwell, 2021-10-20:
if you need screenshots of the device software for features i can send them soon.

MaxHarwell, 2021-10-20:
the dropbox link should be working now if you didnt get it before the link died, please let me know of any updates.

MaxHarwell, 2021-11-17:
> > the dropbox link should be working now if you didnt get it before the link died, please let me know of any updates.
> 
> Yes the droppbox link worked. As i am currently quite busy with work i can't promise you when exactly, but i will keep you updated as soon as possible.

Its been about a month, any updates?

dokutan, 2021-11-17:
I'm sorry for not keeping you updated, the current situation is as follows:

The M686 seems to be similar to the M913 (same vendor id, similar captured data). In commit 8066172567d8d7bfe0b74a911c4367da3f697afd i added an experimantal backend for the M686 by copying the existing one from the M913.
This means you can now try out a few settings that are already implemented (please use the M913 example config for now), reporting which implemented features (don't) work would be helpful.

Sadly i don't fully understand the protocol of these mice and am encountering a lot of weird stuff (on the M913 at least), which further slows down the progress. My current approach is to work on the protocol primarily by examining the M913 as i have access to one, then implement the features for both mice, testing and maybe modifying would still be required for the M686. 
However, if you are interested in reverse engineering the protocol, i would definitely appreciate the help and gained speed.

MaxHarwell, 2021-11-18:
i will try to look into messing with the 913 settings and seeing how well it works with the 686, this will take some time to figure out as i'm not the most knowledgeable on Linux, as I usually follow instructions and google stuff about issues i run into a lot. so please bear with me.

maxtimbo, 2021-12-29:
Hello, 

I just grabbed one of these mice, plugged it in, installed this software, and followed this thread to here. 
Note that the output of lsudb is different for me, though:

```
Bus 004 Device 003: ID 25a7:fa35 Areson Technology Corp 2.4G Dual Mode Mouse
Bus 008 Device 005: ID 25a7:fa34 Areson Technology Corp 2.4G Wireless Receiver
```

When I used the M913 config, everything was wacky. I used the output of `mouse_m908 -R -` and compared it the the `example_m913.ini` and matched the buttons as such:

```
[profile1]

lightmode=rainbow
color=00ff00
brightness=50

dpi1=1000
dpi2=3000
dpi3=4000
dpi4=6000
dpi5=16000

button_left=dpi-cycle
button_right=none
button_middle=dpi+
button_fire=dpi-
button_1=left
button_2=right
button_3=middle
button_4=none 
button_5=none 
button_6=none 
button_7=none 
button_8=none 
button_9=none 
button_10=none 
button_11=none 
button_12=none 
```
After that I tried messing around with the buttons to see which is which:

```
button_left=dpi-cycle
button_right=none
button_middle=dpi+
button_fire=dpi-
button_1=left
button_2=right
button_3=middle
button_4=A
button_5=B
button_6=C
button_7=D
button_8=e
button_9=f
button_10=g
button_11=h
button_12=i
```

That didn't change any of those buttons. I figured I'm probably giving the keys incorrect values. 

I also tried plugging it in. This is the output of `mouse_m908 -R -` after plugging it in:

```
$ mouse_m908 -R -
# Model: 686
[... 87 lines cut ...]
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 02 00 00 00 01 01 00 00 00 00 00 00 00 00 00 48 
09 04 00 00 00 02 08 01 00 00 00 00 00 00 00 00 3d 
```
I'd like to be able to map the three buttons at least for `backwards` and `forwards`. Anything else would also be cool. But that'd be a start. Where should I go from here?

maxtimbo, 2021-12-30:
Quick update:

I fired up a windows VM and installed the software so I could set the side buttons. I then ran `mouse_m908 -R -`:

```
$ mouse_m908 -R -
# Model: 686
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# Note: this feature is incomplete for the m686.

# Battery level: 90 %

# The m686 has two profiles that can be switched using the 'mode switch' button on the bottom of the mouse.
# Reading the settings can only be done for the active profile, therefore only profile1 is used in this config.
[profile1]

report_rate=1000

# DPI settings
# Currently active DPI level: 1
dpi1=1000
dpi2=3000
dpi3=4000
dpi4=6000
dpi5=7500

# Button mapping
button_left=none
button_right=none
button_middle=dpi+
button_fire=dpi-
button_1=left
button_2=right
button_3=middle
button_4=none
button_5=forward
button_6=unknown, please report as bug:  9  0  0  4c
button_7=backward
button_8=none
button_9=none
button_10=none
button_11=none
button_12=none
```
Note* `button_6` is Button 8 and it's disabled in the sofware.

dokutan, 2021-12-30:
@maxtimbo Thanks, i have updated the button names and added a config file example for the M686 to mostly match your description. The difference is that the buttons are identified by their name, not by their number.

Regarding your first comment, there are still a few open issues:
- Mapping a button as a keyboard key is currently not implemented
- Based on my experience with the M913: Sometimes reading the button mapping fails, however retrying always worked for me.

maxtimbo, 2021-12-31:
hmm. well, i'd like macro/key press support. Can I get you one of these mice to hasten the support for m686 mice?

dokutan, 2021-12-31:
> Can I get you one of these mice to hasten the support for m686 mice?

I appreciate the offer, but would like to currently decline for two reasons:
- as i have access to a M913 which uses the same protocol i see no large benefit in having the M686 as well, therefore
- in case of the M686 the limiting factor is not a lack of hardware, but a lack of free time and a protocol that is considerably more complex to implement.

> i'd like macro/key press support

I have had another look at implementing support for keyboard keys, and now have a decent understanding of this part of the protocol (https://github.com/dokutan/mouse_m908/blob/master/documentation/m913-key-press.txt). The biggest problem is that some form of checksum gets used which is extremely annoying.
The best thing you could do to get key press support is to help with the reverse engineering of the protocol.

maxtimbo, 2021-12-31:
> help with the reverse engineering of the protocol.

How do I do that? Is that the wireshark bit you mentioned earlier? Is there an easier way to get (i assume are) hex dumps? And then, what'll you want me to do? Program some macros and key presses, and just get the data being sent to the mouse?

dokutan, 2021-12-31:
The idea is to use wireshark to capture the data sent by the official software (or this software for debugging) for different values, then compare the data. I found it useful to export the packets as json, then run
```
#!/usr/bin/env sh
grep usb.data_fragment "$1" | cut -d \" -f4 > "$1".data
```
on the exported json to obtain a format that is suitable for ``diff``. Experimenting with the packets in the source code (include/m6868/data.cpp, mostly to confirm your assumptions) can be helpful as well.

The best thing woud be if you could find out how the checksum in the [key press](https://github.com/dokutan/mouse_m908/blob/master/documentation/m913-key-press.txt) part of the protocol works.

maxtimbo, 2022-01-01:
Where can I dump all this data? I went through and captured every alphabet letter as well as all possible modifier combinations. You can only have modifiers assigned at a time per button. I noticed that some combinations send two or three packets per key. I have all the json files trimmed using that bask script. There are 15 files in total. I could just copy-pasta all these since it's typically the first 7 packets that do stuff. Also, software doesn't allow you to not have a left click assigned. Here are my notes while I was doing all this:

|filename|b1|b2|b3|b4|b5|b6|b7|b8|
|---|---|---|---|---|---|---|---|---|
|s0|leftclick|a|b|c|d|e|f|g|
|s1|leftclick|h|i|j|k|l|m|n|
|s2|leftclick|o|p|q|r|s|t|u|
|s3|leftclick|v|w|x|y|z|shift+a|ctrl+a|
|s4|leftclick|alt+a|win+a|shift+ctrl+a|ctrl+alt+a|alt+win+a|shift+alt+a|alt+win+a|

Changing only button 6 as follows:

- s5 - shift+a
- s6 - ctrl+a
- s7 - alt+a
- s8 - win+a
- s9 - shift+ctrl+a
- s10 - shift+alt+a
- s11 - shift+win+a
- s12 - ctrl+alt+a
- s13 - ctrl+win+a
- s14 - alt+win+a

You cannot set more than two modifiers.

dokutan, 2022-01-02:
> I went through and captured every alphabet letter as well as all possible modifier combinations.

Thanks, that sounds promising.

> Where can I dump all this data?

I think the easiest option would be for you to put everything in in a directory under ``documentation`` and create a pull request.

maxtimbo, 2022-01-03:
Done. Also, I'm far more versed in python than I am in C++, but I have written C++ in the past (for an entry college course a few years back). That being said, I was looking at the bit that writes the config file to the mouse, I believe it's rd_mouse.cpp. I was thinking about the behavior when a wireless mouse is plugged in with both USB and the wireless donglem how it's very slow and sometimes returns erroneous data.  

So around line 93, you get the vid and pid of the device. However, when it's plugged in, those would both return two values, correct? 

Continuing to read the rd_mouse.h file, it looks like the mouse must first be detached from the kernal driver before you can get/set values. So if there are two sets of vid/pid, why not detach both, write to one, and then reattach both?

I might be way off base here, so apologies if this only confuses things.

dokutan, 2022-01-03:
> So around line 93, you get the vid and pid of the device. However, when it's plugged in, those would both return two values, correct? (If both wired and wireless interfaces are present.)

What i am doing is iterating over all usb devices, for each device iterate over all mice and check if their vid and pid match. Getting the vid/pid ( e.g. ``uint16_t vid = descriptor.idVendor;``) of a device will always return exactly one value. When multiple devices/PIDs are present the last one gets used.

> Continuing to read the rd_mouse.h file, it looks like the mouse must first be detached from the kernal driver before you can get/set values. So if there are two sets of vid/pid, why not detach both, write to one, and then reattach both?

There is no need to detach both, only the driver for the actually used one needs to be detached.

The best way to prefer the wired over the wireless connection i currenty see is to remove or skip the wirelsess pid if the other one is present. Better suggestions or implementations are definitely welcome.

dokutan, 2022-01-03:
@maxtimbo Mapping a button as a single key a-z is implemented with 877310edf0dcf3347768bbabb581720c08d07315, i will now work on the remaining keyboard keys and key combinations. It would be great if you could test the current implementation with the M686.

maxtimbo, 2022-01-04:
@dokutan Superb. I tried a handful of a-z as well as some other keys such as F1, slash, etc. All seem to be working. "Ro" didn't do anything (I was curious to see what it generated, It's likely that I don't have the font/language to handle that). 

Also worth noting, when I `mouse_m908 -R -` after setting any keys, it reports `default` despite the button giving me the correct key press.

How can I help further with modifiers?

dokutan, 2022-01-04:
@maxtimbo

> Also worth noting, when I mouse_m908 -R - after setting any keys, it reports default despite the button giving me the correct key press.

I haven't looked into reading the keyboard keys, but plan to do that after the modifiers are working.

> How can I help further with modifiers?

Basically the same as with individual keys, capture data and use it to figure out how the protocol works.

Here is what i already know: 
- When modifiers are used, two packets are sent. It would help to have a file for each button, containing only these two packets.
- second_packet[4] = first_packet[4] + 0xa

epicfacethe3rd, 2023-08-25:
i have no idea how useful i can be here, but i've also got an m686 i'm willing to help use to improve the project!

epicfacethe3rd, 2023-08-25:
minor note, for me at least, the forwards/backwards buttons seem to not be working?

epicfacethe3rd, 2023-08-25:
a few things i've noticed in XEV:
some of the side buttons are currently showing up as keys?
```
KeyPress event, serial 29, synthetic NO, window 0x5200001,
    root 0x6c2, subw 0x0, time 3583670, (-136,4), root:(753,485),
    state 0x0, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
middle button
```
ButtonPress event, serial 25, synthetic NO, window 0x3800001,
    root 0x6c2, subw 0x0, time 306901, (75,80), root:(964,561),
    state 0x0, button 9, same_screen YES
```
back button
```
KeyPress event, serial 28, synthetic NO, window 0x5200001,
    root 0x6c2, subw 0x0, time 3660326, (-578,-164), root:(311,317),
    state 0x0, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

epicfacethe3rd, 2023-08-27:
nvm - for some reason this was what they were configured to out of the box. put it on a windows machine, and both forwards and backwards now simply show up as `keyboard_key` in the config.ini file when exported. Also, after my most recent flash with windows software, the program will no longer dump the config.ini file

CodexHere, 2024-07-10:
I just got one of these mice and was curious if there is anything that'd help get RGB settings applicable? I noticed in the example config it says it's not supported, and sure enough I didn't notice it take effect.

Dying to change the color on the fly (it helps me figure out my profile).