dokutan/mouse_m908 #33: Support Redragon M901 Perdition

venohm69, 2021-11-22:
I have this mouse and would like full support.  Happy to help any way I can.  I am not a full time coder and haven't been for to many years to write driver code, but am happy to supply any information on this mouse that I can or other ways to support this effort.

Thank you,

dokutan, 2021-11-22:
To start adding support would require a capture of the usb communication with the official software and a detailled description of the features the M901 has (number of buttons, minimum/maximum DPI, …).  I suggest that you have a look at the other issues, especially the closed ones, to find further information, as the initial steps are identical between mice.

venohm69, 2021-11-23:
Okay, I'll start looking at them and provide the information.  The windows part may take the longest since I finally made the switch to linux 100% on my main desktop, but I do still have access to windows systems but since they aren't my primary it will slow me down on getting that portion of the information.

dokutan, 2021-11-23:
Thanks for the lsusb output, the product id is different from the one listed in the readme. I have added it to the generic backend with 4cbbb26d4ddbe80b820472639430edc0bc131d78, in case you want to try that out.

> The windows part may take the longest since I finally made the switch to linux 100% on my main desktop, but I do still have access to windows systems but since they aren't my primary it will slow me down on getting that portion of the information.

I don't know if you have considered it, but running windows inside a qemu/kvm virtual machine works well with usb passthrough, at least that is what i do.

venohm69, 2021-11-24:
Just realized I still have the old drive with windows still installed on it.  I still had some Data to move off, so I hadn't formatted it yet.  I've configured the dual boot and will reboot into windows to get the info.  Thankfully I hadn't wiped that drive yet. I hadn't moved my Skyrim saves and mods over yet.  :-D

I'll also give the generic backend a  try with the new entry as well.

Thanks for your work on this.