dokutan/mouse_m908 #35: nearly broke my m607

xipe-totek, 2022-01-20:
oh god, I just read the instructions, found example config file for this mouse, made a couple of changes, tried to apply - and then mouse died. also I got warnings about unknown DPI value (example file said you can specify the DPI by exact value, but obviously that's a lie). lucky me, there is another computer with Windows, and I have revived my mouse by Redragon's soft. I understand that you dont give any guarantee, so I dont want to blame or offense you. I just want to say people - dont use this program if you are not IT-genius.

xipe-totek, 2022-01-21:
looks like something wrong with identification of that mouse. sticker says it is Griffin M607, but lsusb has different opinion

`lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 010: ID 04d9:fc38 Holtek Semiconductor, Inc. Gaming Mouse [Redragon M602-RGB]
Bus 003 Device 002: ID 0408:a060 Quanta Computer, Inc. HD Webcam
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 04ca:3016 Lite-On Technology Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

dokutan, 2022-01-21:
I'm very sorry this software caused problems with your mouse. 
The cause is that for some reason the generic backend was used and not the backend for the M607. Adding ``-M 607`` as a commandline option would enforce using the correct backend
What version of this software did you use (release or did you build form the master branch), maybe your version doesn't have the M607 backend?

> (example file said you can specify the DPI by exact value, but obviously that's a lie)

Actual DPI values should be working with the correct backend.

>  looks like something wrong with identification of that mouse. sticker says it is Griffin M607, but lsusb has different opinion

The device gets identified by the vendor id and product id (04d9:fc38), most likely the M602 has the same ids as the M607 leading to the wrong identification. This shouldn't cause any problems.

dokutan, 2022-01-22:
> I'm using version 3.2-1 from AUR (my distro is Manjaro). for some reason it don't know M607:

Support for the M607 was added after version 3.2. I have made a new release (3.3) and updated the AUR as well.

xipe-totek, 2022-01-22:
thank you, I have updated, and now it recognizes the model correctly. here is new config. I didn't check how it works and I hope you forgive me my cowardise - if I break something, I can hardly fix it. also there is still something strange with DPI values and buttons. for example, button_dpi defined as backward.
[M607_new.txt](https://github.com/dokutan/mouse_m908/files/7917693/M607_new.txt)
this is how it really looks for profile 1 (soft running with WINE and at least correctly reads the buttons). you can compare it with the config.
[image]

on this screenshot it is original button mapping. looks like program dont really reads the buttons but doing something random lol )) 
[image]

xipe-totek, 2022-01-23:
hm, I think it will be good to make some kind of rollcall, to collect statistics on working with various models. so that people better understand what to expect from this program. many had problems, and this is evidenced by issues. but how many people use this program without any troubles?

dokutan, 2022-01-23:
> I didn't check how it works and I hope you forgive me my cowardise - if I break something, I can hardly fix it.

That is understandable, you are not the first person to break your mouse, however as far as i know it has always been fixable with the official software: https://github.com/dokutan/mouse_m908/blob/master/README.md#safety

> there is still something strange with DPI values and buttons. for example, button_dpi defined as backward.

To fix the buttons i would need at least a capture of the usb communication with the official software, also it is unclear if this problem affects writing the settings to the mouse as well. Some DPI values are unknown but i could add them if you would
 use the official software to get the correct DPI values.

> so that people better understand what to expect from this program

I hoped the [relevant section in the readme](https://github.com/dokutan/mouse_m908/blob/master/README.md#supported-mice) would make that clear. If that is not the case and you have any ideas for clarification i would be interested in hearing them. As a result of this issue i have changed the status of the M607 from complete' to 'partial'.

I am closing this issue now as the original problem was affecting the generic backend, feel free to open a new one if you want to contibute to the M607 backend or reply to this one if you have anything concerning the current discussion.

closed 2022-01-23 by dokutan