dokutan/mouse_m908 #42: m913: Fire configuration (also

bendlas, 2022-09-20:


dokutan, 2022-09-21:
Thanks, i have added initial support for the fire button in 6f4e9b5c9fdfc43cd241a0a3d8f9e1736a9588bf, i hope this works. If you can provide examples of the bytes for other fire button configurations and the snipe button i can try to add them as well. The same applies to ``Enter``, however that might be more complicated (a button gets set as a keyboard key but the actual keycode is configured somewhere else), so a complete capture of the usb communication might be needed in that case.

bendlas, 2022-10-04:
I tried with `button_fire=fire:3:1` as well as `button_fire=fire`, but it doesn't seem to work, and `-R -` shows `button_fire=none`.

Just wanted to provide my observations for now. I'll ping back here when I managed to get a VM setup and some capture traces.