dokutan/mouse_m908 #45: M801 Sniper Pro Support

studkid, 2022-11-10:
I have a M801 Sniper Pro mouse and was wonder if it would be possible to add support for it.
From a `lsusb` I got
```Bus 001 Device 008: ID 25a7:fa0a Areson Technology Corp 2.4G Dual Mode Mouse```
Though I noticed this is different from the 04d9 that seems to be expected so I'm unsure if that causes any issues.  I can try to provide any other information if necessary.

Thank you!

dokutan, 2023-11-03:
I'm really sorry for ignoring this issue for a year, i completely forgot to reply.

The PID of the M801 is the same as the PID of other wireless mice (M913, M686) which use a different and incompatible protocol as the other mice. Unless someone reverse engineers that protocol, supporting these mice is impossible.

trajano, 2024-01-03:
Maybe the README can be updated to include this one as unsupported.

dokutan, 2024-01-07:
> Maybe the README can be updated to include this one as unsupported.

Done in f35e93ca8a8d961d745f891ef62ae4e665c6d585, i'm closing this issue for now.

closed 2024-01-07 by dokutan