dokutan/mouse_m908 #49: possible to assign other mouse functions to m908.

amper5and, 2023-02-18:
I would like to assign mouse right/left tilt click function to a key on the m908 (tilt click is when the scroll wheel tilts to the right/left on some mice and actually registers a click - usually for scrolling horizontally). The m908 doesn't actually have native tilt click functionality but is it possible to use this to program the m908 to send a tilt click when pressing a key?

dokutan, 2023-02-18:
> is it possible to use this to program the m908 to send a tilt click when pressing a key?

Unfortunately not, as the functionality is limited by the firmware.

I suggest mapping the keys to some unused keyboard keys (e.g. F13-F24 or some media controls) on the mouse, then using any of the uinput based tools to remap those to your desired events in userspace.

closed 2023-02-18 by dokutan

amper5and, 2023-02-18:
That's what I suspected. Thank you for the reply and for this project.