dokutan/mouse_m908 #52: Package not building (arch and kde)

MingTheNoob, 2023-06-10:
When trying to install the package thru the [snapshot](https://aur.archlinux.org/cgit/aur.git/snapshot/mouse_m908.tar.gz) or with the PKGBUILD from cloning the repo, the build process was stopped due to an error. Here is the related output:

```cpp
In file included from include/rd_mouse_wireless.cpp:19:
include/rd_mouse_wireless.h:35:58: error: ‘uint8_t’ was not declared in this scope
35 |                 static std::map< std::string, std::array<uint8_t, 4> > _c_keycodes;
|                                                          ^~~~~~~
include/rd_mouse_wireless.h:26:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
25 | #include <vector>
+++ |+#include <cstdint>
26 |
include/rd_mouse_wireless.h:35:68: error: template argument 1 is invalid
35 |                 static std::map< std::string, std::array<uint8_t, 4> > _c_keycodes;
|                                                                    ^
include/rd_mouse_wireless.h:35:70: error: template argument 2 is invalid
35 |                 static std::map< std::string, std::array<uint8_t, 4> > _c_keycodes;
|                                                                      ^
include/rd_mouse_wireless.h:35:70: error: template argument 4 is invalid
include/rd_mouse_wireless.h:38:24: error: ‘uint8_t’ does not name a type
38 |                 static uint8_t _i_keyboard_key_checksum(uint8_t scancode);
|                        ^~~~~~~
include/rd_mouse_wireless.h:38:24: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.h:41:35: error: ‘uint8_t’ was not declared in this scope
41 |                 static std::array<uint8_t, 17> _c_data_button_as_keyboard_key;
|                                   ^~~~~~~
include/rd_mouse_wireless.h:41:35: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.h:41:46: error: template argument 1 is invalid
41 |                 static std::array<uint8_t, 17> _c_data_button_as_keyboard_key;
|                                              ^
include/rd_mouse_wireless.cpp:21:35: error: ‘uint8_t’ was not declared in this scope
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                   ^~~~~~~
include/rd_mouse_wireless.cpp:20:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
19 | #include "rd_mouse_wireless.h"
+++ |+#include <cstdint>
20 |
include/rd_mouse_wireless.cpp:21:45: error: template argument 1 is invalid
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                             ^
include/rd_mouse_wireless.cpp:21:47: error: template argument 2 is invalid
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                               ^
include/rd_mouse_wireless.cpp:21:47: error: template argument 4 is invalid
include/rd_mouse_wireless.cpp:21:49: error: scalar object ‘rd_mouse_wireless::_c_keycodes’ requires one element in initializer
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                                 ^~~~~~~~~~~~~~~~~
include/rd_mouse_wireless.cpp:37:1: error: ‘uint8_t’ does not name a type
37 | uint8_t rd_mouse_wireless::_i_keyboard_key_checksum(uint8_t scancode){
| ^~~~~~~
include/rd_mouse_wireless.cpp:37:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.cpp:41:12: error: ‘uint8_t’ was not declared in this scope
41 | std::array<uint8_t, 17> rd_mouse_wireless::_c_data_button_as_keyboard_key = {
|            ^~~~~~~
include/rd_mouse_wireless.cpp:41:12: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.cpp:41:23: error: template argument 1 is invalid
41 | std::array<uint8_t, 17> rd_mouse_wireless::_c_data_button_as_keyboard_key = {
|                       ^
include/rd_mouse_wireless.cpp:41:25: error: scalar object ‘rd_mouse_wireless::_c_data_button_as_keyboard_key’ requires one element in initializer
41 | std::array<uint8_t, 17> rd_mouse_wireless::_c_data_button_as_keyboard_key = {
|                         ^~~~~~~~~~~~~~~~~
make: *** [makefile:114: rd_mouse_wireless.o] Error 1
==> ERROR: A failure occurred in build().
Aborting...
```

when using paru to install the package the following error output occurs:
```cpp
In file included from include/rd_mouse_wireless.cpp:19:
include/rd_mouse_wireless.h:35:58: error: ‘uint8_t’ was not declared in this scope
35 |                 static std::map< std::string, std::array<uint8_t, 4> > _c_keycodes;
|                                                          ^~~~~~~
include/rd_mouse_wireless.h:26:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
25 | #include <vector>
+++ |+#include <cstdint>
26 |
include/rd_mouse_wireless.h:35:68: error: template argument 1 is invalid
35 |                 static std::map< std::string, std::array<uint8_t, 4> > _c_keycodes;
|                                                                    ^
include/rd_mouse_wireless.h:35:70: error: template argument 2 is invalid
35 |                 static std::map< std::string, std::array<uint8_t, 4> > _c_keycodes;
|                                                                      ^
include/rd_mouse_wireless.h:35:70: error: template argument 4 is invalid
include/rd_mouse_wireless.h:38:24: error: ‘uint8_t’ does not name a type
38 |                 static uint8_t _i_keyboard_key_checksum(uint8_t scancode);
|                        ^~~~~~~
include/rd_mouse_wireless.h:38:24: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.h:41:35: error: ‘uint8_t’ was not declared in this scope
41 |                 static std::array<uint8_t, 17> _c_data_button_as_keyboard_key;
|                                   ^~~~~~~
include/rd_mouse_wireless.h:41:35: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.h:41:46: error: template argument 1 is invalid
41 |                 static std::array<uint8_t, 17> _c_data_button_as_keyboard_key;
|                                              ^
include/rd_mouse_wireless.cpp:21:35: error: ‘uint8_t’ was not declared in this scope
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                   ^~~~~~~
include/rd_mouse_wireless.cpp:20:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
19 | #include "rd_mouse_wireless.h"
+++ |+#include <cstdint>
20 |
include/rd_mouse_wireless.cpp:21:45: error: template argument 1 is invalid
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                             ^
include/rd_mouse_wireless.cpp:21:47: error: template argument 2 is invalid
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                               ^
include/rd_mouse_wireless.cpp:21:47: error: template argument 4 is invalid
include/rd_mouse_wireless.cpp:21:49: error: scalar object ‘rd_mouse_wireless::_c_keycodes’ requires one element in initializer
21 | std::map< std::string, std::array<uint8_t, 4> > rd_mouse_wireless::_c_keycodes = {
|                                                 ^~~~~~~~~~~~~~~~~
include/rd_mouse_wireless.cpp:37:1: error: ‘uint8_t’ does not name a type
37 | uint8_t rd_mouse_wireless::_i_keyboard_key_checksum(uint8_t scancode){
| ^~~~~~~
include/rd_mouse_wireless.cpp:37:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.cpp:41:12: error: ‘uint8_t’ was not declared in this scope
41 | std::array<uint8_t, 17> rd_mouse_wireless::_c_data_button_as_keyboard_key = {
|            ^~~~~~~
include/rd_mouse_wireless.cpp:41:12: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
include/rd_mouse_wireless.cpp:41:23: error: template argument 1 is invalid
41 | std::array<uint8_t, 17> rd_mouse_wireless::_c_data_button_as_keyboard_key = {
|                       ^
include/rd_mouse_wireless.cpp:41:25: error: scalar object ‘rd_mouse_wireless::_c_data_button_as_keyboard_key’ requires one element in initializer
41 | std::array<uint8_t, 17> rd_mouse_wireless::_c_data_button_as_keyboard_key = {
|                         ^~~~~~~~~~~~~~~~~
make: *** [makefile:114: rd_mouse_wireless.o] Error 1
==> ERROR: A failure occurred in build().
Aborting...
error: failed to build 'mouse_m908-3.3-1':
error: packages failed to build: mouse_m908-3.3-1
```
When trying to install with pacman, it says package not found

Techwizz-somboo, 2023-06-10:
In `src/mouse_m908-3.3/include/rd_mouse_wireless.cpp` and `src/mouse_m908-3.3/include/rd_mouse_wireless.cpp` add `#include <cstdint>` with the other `#include` statements.

Then use `makepkg -si -e`

The `-e` ensures it won't extract the sources on top of your modified sources.

Techwizz-somboo, 2023-06-10:
PS, this has already been fixed upstream. The AUR package is outdated.

MingTheNoob, 2023-06-10:
Following your instructions worked except makepkg was still giving the same error. i used cmake instead and it built and installed fine. thank you

closed 2023-06-10 by MingTheNoob