dokutan/mouse_m908 #54: M908 Fire Button

rogeriofrsouza, 2023-07-01:
I'm experiencing some issues with the M908 fire button in Manjaro Linux, I got the AUR package 3.4-1.

The button isn't working properly here, most of the times the effect of double/triple left mouse click isn't occurring.
This is my configuration:

```
# Model: 908
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# Note: reading the scrollspeed is not supported by the mouse.

# Currently active profile: 1

[profile1]

# LED settings
color=ffffff
brightness=1
speed=2
lightmode=static
report_rate=500

# DPI settings
# Active dpi level for this profile: 1
dpi1_enable=0
dpi1=500
dpi2_enable=0
dpi2=1000
dpi3_enable=0
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:3:1
button_dpi_up=dpi+
button_dpi_down=dpi-
button_1=F1
button_2=F2
button_3=F3
button_4=F4
button_5=F5
button_6=F6
button_7=F7
button_8=F8
button_9=F9
button_10=F10
button_11=F11
button_12=F12
scroll_up=scroll_up
scroll_down=scroll_down
```

dokutan, 2023-07-01:
The fire button is at least configured as intended, as ``mouse_m908 -R`` shows, and using the fire button works on my M908. This leaves a few possibilities i can think of:
- your mouse has a slightly different firmware, there isn't much i could do in that case
- the events are sent to fast for the application you are using to test, try testing the fire button with ``evtest``
- the delay is to small for dsomething else, try e.g.: ``fire:mouse_left:3:10``

rogeriofrsouza, 2023-07-01:
Oh, just changed the delay for a higher value and it's working fine now.

I'm using : ```button_fire=fire:mouse_left:3:2```

Thank you so much!

closed 2023-07-01 by rogeriofrsouza