dokutan/mouse_m908 #55: M910-K Not working

Marcel-Rodekamp, 2023-07-09:
Hi, 
I got a Redragon M910-K mouse and applied the program using the example_generic file. After applying that the mouse can only go left and right but not up and down anymore. What did I do wrong? Is there a fix so can one revert the changes?

dokutan, 2023-07-09:
> What did I do wrong?

You did nothing wrong when using the software, your only mistake might be not reading the safety section in the readme.

> Is there a fix so can one revert the changes?

Your best chance is using the official software, e.g. in a Windows VM with USB passthrough.

Marcel-Rodekamp, 2023-07-09:
Yes, I have read this but miss understood. For some reason, I thought M602A-RGB was some mouse type... Alright. I will try your fix later. Is there any information that I can give you to help fix this for the future?

dokutan, 2023-07-10:
You could capture the communication with the official software using wireshark and give a detailed description of the mouse (configurable buttons, DPI range), this would allow the creation of a dedicated backend for the M910. There are other issues with details regarding that process, e.g. #2 . Please be aware that adding a backend can be lengthy, potentially not successful and requires testing with access to the hardware.