dokutan/mouse_m908 #56: Change backlight color of REDRAGON M910-K

ricciolino, 2023-08-06:
Hi,

I would only like to change backlight color of my REDRAGON M910-K Gaming mouse. No other special configurations, macros, additive profiles, etc. .
Can I use this software to do this?
Which information I have to post here to understand if this is feasible?
Please, guide me to accomplish this task.

Thanks :)

dokutan, 2023-08-06:
> Can I use this software to do this? 

It might be possible already with the generic backend, but please be aware of the [risks](https://github.com/dokutan/mouse_m908#safety) of this backend.

Adding a dedicated backend for the M910-K that can only change the color shouldn't be too difficult (at least much easier than supporting all features), but will require data collection and testing on your end.

I would need the output of `` lsusb``, and packet captures of the usb communication with the official software. For that you would need to install the official software on windows (possibly a windows vm with usb passthrough), and use wireshark to capture the data sent while changing the color. A few individual captures for different colors should be enough to get started.

Feel free to ask if you have any other questions.

ricciolino, 2023-08-07:
Thanks for your reply @dokutan !

I read the backend risks. As far as understand I could give it a try anyway.... since (in the worst case) I simply could restore functioning situation of my mouse by using REDRAGON original Software under Windows, right?
If it is like this, could you provide me detailed instruction on what I should do to only try changing RGB backlight of my mouse, using default backend of this software ?

If this solution does not work for me, I will proceed with data collection for new dedicated backend implementation for M910-K.

dokutan, 2023-08-07:
>  I simply could restore functioning situation of my mouse by using REDRAGON original Software under Windows, right?

That worked in all cases that i know of. However I can't give you a guarantee that it will work for you (there could be bugs in the firmware or official software).

Unfortunately it seems impossible to change only a single setting like the RGB color on these mice. Even when you change only one setting using the official software, all settings are sent to the mouse. Therefore changing the color will always affect every setting. The only way to get around that is to read all settings from the mouse first, then sent them back unchanged except for the color. Reading the settings is possible using this software but requires a fully working backend to be useful. When writing a minal backend for the M910-K, this would mean resetting everything else to the default settings (the settings used when capturing the usb data) when changing the color.

ricciolino, 2023-08-07:
So what you suggest to me?