dokutan/mouse_m908 #58: m602-1

BOT-Neil, 2023-08-29:
its not detected 
Bus 001 Device 007: ID 258a:1007 SINOWEALTH Game Mouse
pls help

BOT-Neil, 2023-08-30:
I added https://github.com/dokutan/mouse_m908/commit/04dbee7fdde7b1f1ca0005567fb1619ff62f16ab when i try "mouse_m908 -R config.ini"  the mouse becomes unresponsive and requires a restart.

dokutan, 2023-09-03:
The different vendor id and the results of your testing suggest to me that this mouse uses a different protocol than the other mice. This would make support very unlikely, unless someone owning this mouse performs some significant reverse engineering. Therefore i am closing this issue for now, if you or someone else has new information, please reopen it.

closed 2023-09-03 by dokutan