dokutan/mouse_m908 #60: 908 i can readout but not write config

jay-76, 2023-10-14:
hi there
my issue is, i can easy readout my mouse via [sudo mouse_m908 -R -]
it uses the one i made yesterday in a windows vm (just to preset the mouse,before i try my linux parts)
i tried the -c and the -m function and even a -c /path/to/the .ini
the mouse always flashes as it would write it,but it doesnt do so

any suggestions?

dokutan, 2023-11-03:
Unfortunately there is not much i can do with this information, i personally used this program with an M908 without this problem. I suspect that either your mouse has an incompatible firmware (the vendor and product id are identical to mine) or there is a bug in the USB handling. You could try an older version of mouse_m908, in case a bug was introduced at some point.

OWL4C, 2025-06-20:
Have the same issues, whenever i am writing to the mouse i end up with the default config (when retrieved with -R after -c), identifiable by the solid white color of the leds. #71 seems similar. I can still change profile by command. Using debian 12. Seemingly, the problem lies in the write, either it is writing the default config, or it is writing some broken stuff and the mouse failsafes into the default?

OWL4C, 2025-06-20:
I just tried with version 1.6, and it seems to work fine. I am guessing somewhere between current branch and some release there was a breaking change. Since 1.6 works fine, i won't dig in further but it should be trivial to bisect which version is broken and then you can look at the diffs to figure out why.

OWL4C, 2025-06-20:
Okay i actually did have to try out more versions. v2.X had some weird issues with dpi reporting, but v3.2 works perfectly with dpi, leds and everything else i've tried. So download the v3.2 archive, unzip, make, and use that binary directly and it seems to work fine.

dokutan, 2025-06-22:
I tried the current commit (56c46e4318f8c475ce04eef81f10d1c062cce96f) on my M908 (04d9:fc4d) and had no problems writing the example config (and the example config with changes values), so your mouse likely has a different firmware version.

Between 18b7826f9ed924c7b43f33840ed6e434ce549463 (v3.2) and 56c46e4318f8c475ce04eef81f10d1c062cce96f are no obvious commits that could have caused this, it would be great if you could `git bisect` this.