dokutan/mouse_m908 #61: Control RGB Impact M908

cleberabreu87, 2023-11-02:
Hello!

Congratulations on your work with this repo.

I am interested in purchasing a Redragon Impact M908 mouse and would like to know if your solution has a front-end to be able to configure the mouse's RGB?

dokutan, 2023-11-03:
I don't know what exactly you mean by frontend, this program does support configuring the M908. I personally tested this, as i used an M908 for some time, however this is no guarantee that your M908 would be compatible (see for example #60).
If you mean a graphical frontend, then the answer is no, the only implemented interface is the config file.

cleberabreu87, 2023-11-03:
Hi, thanks for the reply.

I actually meant a graphical interface.

Still in time, could you tell me where I can see how to configure the mouse's RGB, as I didn't find it or does this repo only deal with macros?

dokutan, 2023-11-03:
You can configure the RGB by editing a config file and updating the settings by running `mouse_m908 -c path/to/your/config.ini`. Have a look at the example config at https://github.com/dokutan/mouse_m908/blob/master/examples/example_m908.ini for details.

dokutan, 2023-12-31:
I'm closing this as your questions seem to be answered for now – feel free to open a new issue if there is anything else.

closed 2023-12-31 by dokutan