dokutan/mouse_m908 #63: M601-RGB Support

so1der, 2023-12-12:
Hello there! I couldn't reopen [Issue #58](https://github.com/dokutan/mouse_m908/issues/58), thats why I started new one.
I have a M601-RGB mouse (model Centrophorus2), with same VID:PID as in M602-1 from issue I mentioned above.
```
Bus 001 Device 006: ID 258a:1007 SINOWEALTH Wired Gaming Mouse
```
I tried to reverse engineer its protocol, by capturing packages from original software in wireshark, starting with change of mouse mode, and I think I did gather some info, but because of my lack of knowledge in C/USB, I couldn't check it.
To change Mouse Mode, software sends package by USBHID protocol to Endpoint 0x00 OUT:
```
bmRequestType: 0x21
bRequest: SET_REPORT (0x09)
wValue: 0x0305
wIndex: 1
wLength: 6
Data Fragment: 05 02 01 00 00 00
```
01 byte in this case response for mode selection. Mouse have 2 modes, so values are 01 or 02. I tried to send this date myself, but faced numerous problems I couldn't resolve. Thats why Im asking for Your help. Maybe I can create .pcap file, where I will capture all main packages, and describe in detail, what cause those packages, and provide screenshots of original software? Or maybe something else. Thats, of course, if You want to work on supporting such mouse.

dokutan, 2023-12-13:
I will try to implement a backend for this mouse if you provide the required information and are willing to test the changes. However i can't guarantee that i will be successfull at implementing every feature (for example some mice use checksums which makes it really difficult to support them).

Ideally you would add screenshots/a description of the available features, especially the buttons and add one .pcap file for every feature and different setting (creating many pcap files makes it much easier to compare them and not confuse different parts), it would also help to add one pcap file for resetting everything to the default settings if such an option exists in the official software.

so1der, 2023-12-14:
Very glad to hear it! Of course you can put diagram in documentation, I dont mind at all, too bad that quality of diagram is faaar from good. But maybe I will try to re-draw diagram, when I'll have free moment and inspiration :D But for now - feel free to use this diagram as You wish.

I recompile new version of Your software, compilation ended without errors. Commands for mode changing runs without errors as well. But unfortunately mouse didnt responde at all :( I tried with and without 'sudo'. So I provide You packages that were cause by following commands:
```
sudo mouse_m908 -M 601 -p 1
sudo mouse_m908 -M 601 -p 2
```
[m601_mode_selection_packages_linux.zip](https://github.com/dokutan/mouse_m908/files/13674442/m601_mode_selection_packages_linux.zip)

dokutan, 2023-12-14:
Thanks for the new captures, i have tried a few things but unfortunately was unable to send the correct data using libusb. However i was able to get close using the hidapi library from python (i also tried the C version without immediate success):
```py
import hid, time

try:
    m601_profile = 1 # set this to 1 or 2
    h = hid.device()
    h.open(0x258a, 0x1007)
    h.set_nonblocking(1)
    h.send_feature_report([5, 2, m601_profile, 0, 0, 0])
    time.sleep(0.05)
    h.close()

except IOError as ex:
    print(ex)
```
If that works, i would suggest to create a new program for the M601 in python, as there would be almost no shared code between the M601 and the other mice.

so1der, 2023-12-14:
Python version would be awesome, I do know this language and could help in writing GUI version or something.
Sadly, this script didn't work :(
I captured this packages as well, and find a little divergence with original software, its different wIndex:
[image]

Here packages if You need them:

[m601_mode_selection_packages_python.zip](https://github.com/dokutan/mouse_m908/files/13676220/m601_mode_selection_packages_python.zip)

I'll try to do my own research, and if I succeed - I'll let You know, but for now, I guess You can close this issue. Thank you!

dokutan, 2023-12-14:
I hope my work was at least slightly useful. Please let me know when you have a working program, i will add a link to the readme.

closed 2023-12-14 by dokutan

so1der, 2023-12-15:
Hello again! I finally managed to send the package correctly! I did it by using Python library called [PyUSB](https://pypi.org/project/pyusb/). I tried to use this library when I only started to reverse engineer protocol. But again, due to my lack of knowledge in USB I didn't know what I was writing. But after all this package analysing stuff, and a couple of videos/articles, I understnd what I was doing wrong. So here is the code that managed to change modes of this Mouse:
``` Python
import usb.core
import usb.util
import sys

VID = 0x258a
PID = 0x1007
modes = [1, 2]
mode = int(input("Select Mouse Mode (1 or 2):"))

if mode not in modes:
    raise ValueError('Wrong Mode. Expected 1-2')

payload = [0x05, 0x02, mode, 0x00, 0x00, 0x00]

# find our device
dev = usb.core.find(idVendor=VID, idProduct=PID)

# was it found?
if dev is None:
    raise ValueError('Device not found')

# detach if busy:
if dev.is_kernel_driver_active(1):
    try:
        dev.detach_kernel_driver(1)
    except usb.core.USBError as e:
        sys.exit("Could not detatch kernel driver from interface({0}): {1}".format(1, str(e)))

#send HID
dev.ctrl_transfer(
    0x21,   # bmRequestType 
    0x09,   # bRequest SET_REPORT
    0x0305, # ReportID:5, ReportType:3 (Feature) 
    1,      # wIndex: 1
    payload # the HID payload as a byte array
)

```
I'll try to write software in Python, and I'll let you know if/when I'm done.

so1der, 2023-12-23:
Hello once again! So, I guess, I finished writing the software. I decided not to write GUI version, at least for now. In current state, it can read settings, write settings, change mouse mode, and dump raw data. By changing settings I mean - usb polling rate, dpi, all related with lighting, and buttons rebinding. Im currently haven't implemented macros and key combinations yet, but hope some day I will. Im also did a little reserch about M601 and came to a conclusion that M601 and M601-RGB may have different protocols, because in M601-RGB pacakges a large part is taken up by setting up lighting colors, meanwhile M601 have only one color - red.  Alas, I have only M601-RGB. But anyway, here is the software:
[M601-RGB](https://github.com/so1der/m601)
Its really shame that this mouse cant be a part of Your software, and have to be a separate piece of the software.

dokutan, 2023-12-31:
Sorry for replying late, congratiulations for completing your software and thanks for the hint about the M601 vs M601-RGB.
I have removed any code for the M601-RGB from this project, instead i have added the M601-RGB with a link to your software to the table of mice in the readme.