dokutan/mouse_m908 #66: Saftey Feature Request

BloodBlight, 2024-01-31:
I was a dummy and fat fingered a command and sent the "example_m908.ini" to my M612.

This ended  with a mouse that would not move left or right.  Ops!

So, I tried re-sending the "generic" example, no help, has a second mouse, captured its config and sent it, no good.

So, found a Windowz box and fixed it there.

Cool.

But on to the feature request.  Add a list of "known good" USB IDs and/or the ID that the configuration was captured from.  Then if the devices ID isn't in that list, prompt the user with a warning before proceeding.

dokutan, 2024-01-31:
I understand how this can be a problem in some cases, but would think that this problem can be avoided by keeping the personal config in an appropriate location (i.e. not in the examples directory). I suppose a config parameter could be added to specify the backend that the config is intended for (and write it when reading the config), then ask the user if this does not match the real backend without breaking every existing config.

As i currently don't have a lot of time to spend on this project, i'm not going to implement this for now, but would merge a pull request if it keeps backwards compatibility with existing configs.

BloodBlight, 2024-01-31:
Understood, I haven't coded c++ in like 20 years so...  :)

Want me to leave it open or close it?

dokutan, 2024-01-31:
Feel free to leave it open, i might implement this when i have more time, or someone else gives it a try.

dokutan, 2025-05-09:
I think a new section like `[options]` would make sense that can replace/override some command line arguments (model, kernel driver, ...). Conflicting command line and config file options would need to be resolved in some meaningful way (at least with a warning).

I have no particular opionions about the implementation details, so i would leave it up to you to come up with something.