dokutan/mouse_m908 #69: Build error

dratyti22, 2024-04-11:
I am using Fedora 39, errors appear in files when using make. 
Make: 
```
c++ -c include/m990chroma/writers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o writers_m990chroma.o
include/m990chroma/writers.cpp: In member function ‘int mouse_m990chroma::write_settings()’:
include/m990chroma/writers.cpp:99:48: warning: iteration 8 invokes undefined behavior [-Waggressive-loop-optimizations]
   99 |                         buffer3[35+(8*i)+j][8] = _s_keymap_data[i][j][0];
include/m990chroma/writers.cpp:98:35: note: within this loop
   98 |                 for( int j = 0; j < 25; j++ ){
      |                                 ~~^~~~
c++ -c include/m990chroma/readers.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o readers_m990chroma.o
include/m990chroma/readers.cpp: In member function ‘int mouse_m990chroma::read_settings()’:
include/m990chroma/readers.cpp:408:51: warning: iteration 8 invokes undefined behavior [-Waggressive-loop-optimizations]
  408 |                         _s_keymap_data[i-1][j][0] = buffer_in3[j+(20*(i-1))][8];
include/m990chroma/readers.cpp:406:35: note: within this loop
  406 |                 for( int j = 0; j < 20; j++ ){
      |                                 ~~^~~~
c++ -c include/generic/constructor.cpp -std=c++17 -Wall -Wextra -O2 `pkg-config --cflags libusb-1.0` -o constructor_generic.o
```
и это приводит к ошибке в make rpm:
```
rpmbuild --buildroot /home/dratyti/Documents/mouse/mouse_m908/rpmbuild/BUILDROOT --define "_topdir /home/dratyti/Documents/mouse/mouse_m908/rpmbuild" -bb mouse_m908.spec
/bin/sh: line 1: rpmbuild: command not found
make: *** [makefile:55: rpm] Error 127
``` 
How can I fix this?

mercster, 2025-03-27:
I don't think those are connected at all.  The top problems are just warnings, not errors.  The second issue is building the RPM, it wants a BUILDROOT that is not there.  I will be building this later so I will let you know.

mercster, 2025-03-27:
OK so 'make rpm' works fine for me in Fedora 41.  If you can't build the RPM it is no problem... just type `make install` instead of `make rpm`.  It will put the program in /usr/bin or /usr/local/bin... you can use the software the same.  Not having RPM does not matter much, because they don't distribute RPMs anyway, and the software is not being actively updated all the time.  It will not clash with other software, RPM is not needed.

dokutan, 2025-11-17:
These warnings are fixed in dcd7c03ab0752faa143437c01802ff5d447b05a4 .

closed 2025-11-17 by dokutan