dokutan/mouse_m908 #74: macros are not doing anything at all

40476, 2024-06-13:
when I write a macro the key does not respond, also the last field of the fire key command does not seem to be doing anything

```ini
# Model: 908
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# Note: reading the scrollspeed is not supported by the mouse.

# Currently active profile: 1

[profile1]

# LED settings
color=ff0000
brightness=3
speed=0
lightmode=reactive
report_rate=500

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=500
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:25:3
button_dpi_up=Backslash
button_dpi_down=led_mode_switch
button_1=1
button_2=2
button_3=macro1
button_4=4
button_5=5
button_6=macro2
button_7=7
button_8=8
button_9=9
button_10=dpi-
button_11=dpi+
button_12=Num_Minus
scroll_up=scroll_up
scroll_down=scroll_down

##########################################################
##########################################################
##########################################################
##########################################################
##########################################################
##########################################################
##########################################################
##########################################################
##########################################################
##########################################################
##########################################################

[profile2]

# LED settings
color=ff0000
brightness=2
speed=0
lightmode=wave
report_rate=500

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=500
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:3:0
button_dpi_up=dpi+
button_dpi_down=dpi-
button_1=1
button_2=2
button_3=3
button_4=4
button_5=5
button_6=6
button_7=7
button_8=8
button_9=9
button_10=0
button_11=Num_Plus
button_12=Num_Minus
scroll_up=scroll_up
scroll_down=scroll_down

[profile3]

# LED settings
color=ff0000
brightness=2
speed=0
lightmode=wave
report_rate=500

# DPI settings
# Active dpi level for this profile: 3
dpi1_enable=1
dpi1=500
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:3:0
button_dpi_up=dpi+
button_dpi_down=dpi-
button_1=1
button_2=2
button_3=3
button_4=4
button_5=5
button_6=6
button_7=7
button_8=8
button_9=9
button_10=0
button_11=Num_Plus
button_12=Num_Minus
scroll_up=scroll_up
scroll_down=scroll_down

[profile4]

# LED settings
color=ff0000
brightness=2
speed=0
lightmode=wave
report_rate=500

# DPI settings
# Active dpi level for this profile: 4
dpi1_enable=1
dpi1=500
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:3:0
button_dpi_up=dpi+
button_dpi_down=dpi-
button_1=1
button_2=2
button_3=3
button_4=4
button_5=5
button_6=6
button_7=7
button_8=8
button_9=9
button_10=0
button_11=Num_Plus
button_12=Num_Minus
scroll_up=scroll_up
scroll_down=scroll_down

[profile5]

# LED settings
color=ff0000
brightness=2
speed=0
lightmode=wave
report_rate=500

# DPI settings
# Active dpi level for this profile: 5
dpi1_enable=1
dpi1=500
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:3:0
button_dpi_up=dpi+
button_dpi_down=dpi-
button_1=1
button_2=2
button_3=3
button_4=4
button_5=5
button_6=6
button_7=7
button_8=8
button_9=9
button_10=0
button_11=Num_Plus
button_12=Num_Minus
scroll_up=scroll_up
scroll_down=scroll_down

# Macros
;## macro1
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left
;# down mouse_left
;# up   mouse_left

;## macro2
;# down mouse_right
;# up   mouse_right
;# down mouse_right
;# up   mouse_right
;# down mouse_right
;# up   mouse_right
;# down mouse_right
;# up   mouse_right
;# down mouse_right
;# up   mouse_right
;# down mouse_right
;# up   mouse_right
;# down mouse_right
;# up   mouse_right
```

dokutan, 2024-06-14:
> when I write a macro the key does not respond

Have you applied the macros using the `-m` option?

> also the last field of the fire key command does not seem to be doing anything

The last field should be the delay between key events, your mouse might have a slightly incompatible firmware if it has no effect.

40476, 2024-06-14:
> Have you applied the macros using the -m option?
Might I ask why that is separate?
>The last field should be the delay between key events, your mouse might have a slightly incompatible firmware if it has no effect.
odd, I didn't know that may have changed since everything else works.

does it need to have tabs or can therer be spaces?

dokutan, 2024-06-14:
>  Might I ask why that is separate?

Macros were added later and originally had to be specified in separate files, now the only reason is to allow macros and the rest of the settings to be changed independently.

> odd, I didn't know that may have changed since everything else works.

It could also have never worked and since i no longer use an M908 i can't easily test it right now. If you want to fix it, pull requests are always welcome.

> does it need to have tabs or can therer be spaces?

It needs a single tab (i didn't bother to write a better parser at the time).

40476, 2024-06-14:
this is what my macros look like, is there something wrong because only one of them partially works

dokutan, 2024-06-14:
The macros look correct, but i can't test them and have no idea why they aren't working. I suggest you set the mouse buttons to some uncommon keyboard keys and try one of the many [input remap utilities](https://wiki.archlinux.org/title/Input_remap_utilities) that can run on Linux.

40476, 2024-06-16:
so apparently I was just doing it wrong, anyway, how to I make the macro repeat automatically?

40476, 2024-06-17:
is there a way to do automatic repeats? I remember the original mouse driver having the so I was just curious.

dokutan, 2024-06-18:
Use one of:

```
;## macro1:while
;# down	mouse_left
;# delay	1
;# up	mouse_left
;# delay	1

```

```
;## macro1:until
;# down	mouse_left
;# delay	1
;# up	mouse_left
;# delay	1
```

```
;## macro1:50
;# down	mouse_left
;# delay	1
;# up	mouse_left
;# delay	1

```

40476, 2024-06-18:
thanks!! 

What is the difference between `while` anbd `until`?

I will be sure to make a pull request and add a line about the repeating if that is okay!

dokutan, 2024-06-18:
- `while` should repeat the macro while the mouse button is pressed 
- `until` should repeat the macro until the mouse button is pressed again
- a number should repeat the macro number times

These are (badly) documented here: https://github.com/dokutan/mouse_m908/blob/master/keymap.md, if you have an idea how to improve this, please go ahead, better documentation is always a good thing.

40476, 2024-06-19:
it is not doing anything for me, perhaps I missed something again?
```ini
# Model: 908
# Configuration created with mouse_m908 -R.
# This configuration can be send to the mouse with mouse_m908 -c.
# Note: reading the scrollspeed is not supported by the mouse.

# Currently active profile: 1

[profile1]

# LED settings
color=00ffff
brightness=2
speed=0
lightmode=reactive
report_rate=500

# DPI settings
# Active dpi level for this profile: 2
dpi1_enable=1
dpi1=500
dpi2_enable=1
dpi2=1000
dpi3_enable=1
dpi3=2000
dpi4_enable=1
dpi4=3000
dpi5_enable=1
dpi5=6200

# Button mapping
button_left=left
button_right=right
button_middle=middle
button_fire=fire:mouse_left:10:2
button_dpi_up=Backslash
button_dpi_down=led_mode_switch
button_1=e
button_2=left
button_3=macro1
button_4=Esc
button_5=right
button_6=macro2
button_7=7
button_8=8
button_9=macro3
button_10=dpi-
button_11=dpi+
button_12=Num_Minus
scroll_up=scroll_up
scroll_down=scroll_down

# Macro definitions
# Each macro (1-15) is defined in its own section, starting with ";## macro"
# Each action is prefixed by ";# "
# A tab ('\t') is used to separate the action from the value
# There are 4 different possible actions:
# down	(key)
# up	(key)
# delay	(time in 10ms)
# move_(left/right/up/down)	(1-120)

;## macro1:while
;# down	mouse_left
;# up	mouse_left
;# delay 30

;## macro2:while
;# down	mouse_right
;# up	mouse_right
;# delay 30
```

dokutan, 2024-06-19:
It looks right, as i have never personally used that feature i am not 100% sure if it ever worked correctly or if this is a bug that was introduced at some time or if your mouse is slightly incompatible.

40476, 2024-06-19:
it was in the wrong spot

40476, 2024-06-19:
you put it right after you call the macro

closed 2024-06-20 by 40476